Ultimaker/Cura #5463: 4.0.0 Issue selecting object on Macbook Pro

ender2nmm, 2019-03-20:
**Application Version**
4.0.0

**Platform**
Macbook Pro OS 10.14.3 (18D109)

**Printer**
Custom Printer - Ender 2

**Steps to Reproduce**
Load STL and attempt to click the object to select. Nothing happens. Need to us CTRL+A to select object. 

**Actual Results**
Nothing

**Expected results**
Object is selected

**Additional Information**

Ghostkeeper, 2019-03-21:
I have this sometimes too (not specific to MacOS though). Zooming in and then clicking not quite in the centre helps, though. Sometimes there is an invisible cube in the centre of the view.

Does zooming in help for you as well?

motorist828, 2019-03-27:
I have win 10, there are also problems with the selection and movement of the model
I have to increase to do it.

dethfactor, 2019-03-27:
Having the same issue on Win 10. Seems to happen once I clear my plate. Left clicking doesn't select any objects. I have to click both mouse buttons to interact with the object or grab the direction arrows. Kinda a pain with the right click dialog popping up. Restarting clears up the issue till the plate is cleared.

Ghostkeeper, 2019-03-29:
Fixed here: https://github.com/Ultimaker/Uranium/commit/1aa4fe1ee5a27ce6789cebfa39a0489e1ac9ad0a

closed 2019-03-29 by Ghostkeeper