Ultimaker/Cura #5464: Backend Connection reset by peer

cyrius451, 2019-03-20:
**Application Version**

3.6.0

**Platform**

Linux 4.20.7-gentoo #1 SMP Wed Feb 20 21:33:46 CET 2019 x86_64 Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz GenuineIntel GNU/Linux

**Printer**
Log :+1: QML debugging is enabled. Only use this in a safe environment.
[MainThread] UM.Resources.__initializeStoragePaths [414]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [425]: Config storage path is /home/cyrius/.config/cura/3.6
[MainThread] UM.Resources.__initializeStoragePaths [433]: Data storage path is /home/cyrius/.local/share/cura/3.6
[MainThread] UM.Resources.__initializeStoragePaths [445]: Cache storage path is /home/cyrius/.cache/cura/3.6
[MainThread] UM.PackageManager.__init__ [49]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager.__init__ [49]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile
[MainThread] UM.Qt.QtApplication.initialize [138]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [144]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [148]: Initializing version upgrade manager ...
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_definition_changes] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_variant] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_material] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_quality] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_quality_changes] added.
[MainThread] UM.PackageManager._loadManagementData [83]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager._loadManagementData [83]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager._loadManagementData [105]: Loaded user packages management file from /home/cyrius/.local/share/cura/3.6/packages.json
[MainThread] UM.PackageManager._saveManagementData [117]: Package management file /home/cyrius/.local/share/cura/3.6/packages.json was saved
[MainThread] UM.Qt.QtApplication.startSplashWindowPhase [162]: Preferences file not found, ignore and use default language 'en_US'
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [92]: Loading plugin configuration file '/home/cyrius/.config/cura/3.6/plugins.json'
2019-03-20 22:01:49,462 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ConsoleLogger
2019-03-20 22:01:49,474 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [80]: Found CuraEngine at: /usr/bin/CuraEngine
2019-03-20 22:01:49,474 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CuraEngineBackend
2019-03-20 22:01:49,476 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FileLogger
2019-03-20 22:01:49,478 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalFileOutputDevice
2019-03-20 22:01:49,479 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CameraTool
2019-03-20 22:01:49,481 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin TranslateTool
2019-03-20 22:01:49,483 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MirrorTool
2019-03-20 22:01:49,486 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin RotateTool
2019-03-20 22:01:49,488 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ScaleTool
2019-03-20 22:01:49,490 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SelectionTool
2019-03-20 22:01:49,491 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SimpleView
2019-03-20 22:01:49,492 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin OBJReader
2019-03-20 22:01:49,495 - WARNING - [MainThread] STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2019-03-20 22:01:49,496 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin STLReader
2019-03-20 22:01:49,498 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin OBJWriter
2019-03-20 22:01:49,499 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin STLWriter
2019-03-20 22:01:49,500 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalContainerProvider
2019-03-20 22:01:49,503 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UpdateChecker
2019-03-20 22:01:49,503 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin ConsoleLogger was already loaded
2019-03-20 22:01:49,504 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin FileLogger was already loaded
2019-03-20 22:01:49,504 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin LocalFileOutputDevice was already loaded
2019-03-20 22:01:49,504 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin CameraTool was already loaded
2019-03-20 22:01:49,505 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-03-20 22:01:49,505 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin TranslateTool was already loaded
2019-03-20 22:01:49,506 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin MirrorTool was already loaded
2019-03-20 22:01:49,507 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin RotateTool was already loaded
2019-03-20 22:01:49,507 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin ScaleTool was already loaded
2019-03-20 22:01:49,507 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin SelectionTool was already loaded
2019-03-20 22:01:49,507 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin SimpleView was already loaded
2019-03-20 22:01:49,507 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin OBJReader was already loaded
2019-03-20 22:01:49,507 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin STLReader was already loaded
2019-03-20 22:01:49,507 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin OBJWriter was already loaded
2019-03-20 22:01:49,508 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin STLWriter was already loaded
2019-03-20 22:01:49,508 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin LocalContainerProvider was already loaded
2019-03-20 22:01:49,508 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin UpdateChecker was already loaded
2019-03-20 22:01:49,508 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin ConsoleLogger was already loaded
2019-03-20 22:01:49,514 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin 3MFWriter
[... 175 lines cut ...]
    self.run()
  File "/usr/lib64/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib64/python3.5/site-packages/UM/Backend/Backend.py", line 163, in _storeStderrToLogThread
    self._backendLog(line)
  File "/usr/lib64/python3.5/site-packages/UM/Backend/Backend.py", line 94, in _backendLog
    Logger.log("d", "[Backend] " + line_str.strip())
  File "/usr/lib64/python3.5/site-packages/UM/Logger.py", line 68, in log
    logger.log(log_type, message)
  File "/usr/lib/uranium/plugins/FileLogger/FileLogger.py", line 44, in log
    self._logger.debug(message)
Message: '[Thread-14] UM.Backend.Backend._backendLog [94]: [Backend] [ERROR] Connection reset by peer'
Arguments: ()
Exception in thread Thread-14:
Traceback (most recent call last):
  File "/usr/lib64/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib64/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib64/python3.5/site-packages/UM/Backend/Backend.py", line 160, in _storeStderrToLogThread
    line = handle.readline()
OSError: [Errno 9] Bad file descriptor

May i ask you if you have an idea about this issue ?
thx

Ghostkeeper, 2019-03-21:
No idea. It's almost like the CuraEngine executable file disappeared. At least we should catch this error so that it doesn't crash Cura but just resets the engine for the next slice.

Ghostkeeper, 2019-03-21:
Maybe a firewall blocked it?

closed 2019-03-21 by Ghostkeeper