Ultimaker/Cura #5466: Cura 4.0 -- Cannot create custom material with 1.75mm filament

thantik, 2019-03-21:
**Application Version**
Cura 4.0

**Platform**
Windows 7 64 bit.

**Printer**
"Custom FFF"

**Steps to Reproduce**
Try to add a new custom material.  Select filament diameter.  Change it to 1.75.  Get warning that it's not compatible with current extruder, even though current extruder is set to 1.75 - select yes that you're sure you want to continue - Cura then deletes the profile outright and refuses to allow you to get filament diameter (might I mention, this happens with seemingly every new version of Cura.)

**Actual Results**
Cura 4.0 refuses to let you create a custom filament with 1.75mm diameter on Custom FFF profile.

**Expected results**
I should have a custom 1.75mm filament to choose from.

Ghostkeeper, 2019-03-21:
I can't reproduce this issue. These are my steps:

* Add a new Custom FFF Printer.
* In the machine settings, go the first (and only) extruder tab.
* Set Compatible Material Diameter to 1.75mm. Close the pop-ups.
* Go to the material manager. It shows all of the 1.75mm materials there (Chromatik, Octofiber, etc).
* Create a new material. Verify that the material has a diameter of 1.75mm.

Seems to work fine like this. It seems that your printer is not set to be compatible 1.75mm filaments, even though you said `even though current extruder is set to 1.75`. Maybe that part somehow failed for you?

thantik, 2019-03-21:
I will take a screencast of the issue later this evening.  I reinstalled Cura 4.0 just to reconfirm that I could reproduce the issue.  I install Cura 4.0 with default settings.  Fresh out of the box.  I select Custom FFF printer.  I click "Add Printer".  I click the extruder tab, click the filament diameter box, and change it to 1.75.  I click Finish.  I then click the middle dropdown in the plater window, and click "Manage Materials".  I click "Create".  I click the filament diameter of the newly created item.  I enter 1.75, I press tab - it asks me if I'm sure I want to continue, because the diameter is not compatible with the current machine.  I click yes.  The box reverts to 2.85mm.  I click it again.  I type in 1.75, and hit tab to get the same warning again.  I click yes.  The new entry for the filament has now disappeared in the left hand window but left the right-side editing window intact.

If I close Cura 4.0 and reopen it, the created profile is either there, or it's not.  Also not edited for the 1.75 that I entered.

I have installed the VC redist that comes with the installer package.  This happened on previous versions of Cura as well.  Windows 7 64 bit.  I do have Python installed on the system as well as other developer tools.  Could these be interfering in any way?

Ghostkeeper, 2019-03-22:
Here's a screen recording of what happens on my side: http://dulek.net/work/custom_175mm.ogv

thantik, 2019-03-22:
Upon trying to create a screen capture of what was going on, I was unable to recreate the issue.  I uninstalled a bunch of stuff prior to attempting the recreation so something must have been interfering.  Apologies.

closed 2019-03-22 by thantik

CCMCAGP, 2019-08-14:
Hi, I have a similar issue that I reproduced both on Cura 4.0 and 4.2.1, both Mac and Windows. 

So I have a default Original Prusa i3 MK3S printer profile, which is defaulted to 1.75mm, and a custom material with a filament diameter of 1.90mm, and I am trying to make Cura acknowledge this new diameter. No matter what I do, in the end, the profile doesn't get updated with the new diameter.

The steps are the following:

- In a printer profile such as Original Prusa i3 > Machine Settings > Extruder 1, change 'Compatible Material Diameter' to 1.9mm, Close. (you can check by going back that the setting is kept at 1.90mm afterwards)
- Create a custom material, change the diameter to 1.90mm, Activate
- In the Profile tab, the activated profile stays set at a diameter of 1.75mm in the tab Extruder 1, whereas it should have been updated when I changed the material (?) 
- When I slice the object and export a g-code, I can see that in the last lines of the g-code that gathers all the printer parameters, nozzle diameter is still set to 1.75mm, and similarly looking at the extrusion values, with some easy calculations I can also identify that the printer still thinks the filament is 1.75mm, leading to over-extrusion. 

As said earlier I have the problem both on Mac and Windows with Cura 4.0 and 4.2.1.

Now, the only way I found to update the profile is very hacky. It consists in basically manually editing the .curaprofile file in a text editor, (can be done this way: https://www.imade3d.com/2018/02/edit-cura-profile-text-editor-power-user-zone/) and then selecting it in Cura. I don't like this solution a lot as I would like to be able to edit easily the filament diameter since I'm always dealing with irregular experiental filaments...

ggossage, 2022-02-13:
I am having a similar issue with Windows 10, Version 4.7.1. I was able to get the material to show back up by loading a project that used the material.