Ultimaker/Cura #5467: Cura 4.0.0 Slicing of model is not correct

chputney, 2019-03-21:
**Application Version**
Cura 4.0.0

**Platform**
Windows 7
[Globe Shade (1).zip](https://github.com/Ultimaker/Cura/files/2991263/Globe.Shade.1.zip)

**Printer**
Wanhao Duplicator i3Plus

**Steps to Reproduce**
Load model Globe Shade (1).stl into Cura.  The view is correct, but when the model is sliced, the cylinder at the bottom of the globe is missing.

[Globe Shade (1).zip](https://github.com/Ultimaker/Cura/files/2991265/Globe.Shade.1.zip)

smartavionics, 2019-03-21:
I think the slicing is correct, the real problem is that the wall in the cylinder at the bottom is so thin, it's not even being printed as a thin wall. Look at it on this image of the model...

[image]

Ghostkeeper, 2019-03-21:
Yeah that sleeve is about 0.018mm thick, so it's like a 20th of your nozzle size and can't be printed accurately.

Not even the Print Thin Walls function can fit anything significant in there because it's fairly close to rounding errors.

chputney, 2019-03-21:
Whoops.  Very sorry I didn't check model more thoroughly.   This issue can be closed.
Charles Putney

closed 2019-03-22 by Ghostkeeper