Ultimaker/Cura #5468: Not remembering save path

regularlabs, 2019-03-21:
Using Cura 4 on OSX.

When opening Cura and trying to save a file (gcode), the location is always set to HD (it was Applications in Cura 3).
[image]

You can't save files to that location and you need to manually select the location to save to.
All fine.

However, after closing Cura and restarting it, it is reset to that default HD location again.

So question is:
Can the default location be set somewhere?
If not, can this issue be fixed, so that it remembers the last save location.

regularlabs, 2019-03-21:
PS: May be connected to #4774, however that concerns the open path.

Ghostkeeper, 2019-03-21:
Seems to be OSX-specific, since it seems to work fine on Ubuntu 18.10 and Windows 8.

regularlabs, 2019-03-21:
Ok, but still a bug...

regularlabs, 2019-03-22:
To make this even more confusing:
- I click on save, it shows me the incorrect HD root path
- I click on cancel
- I click on save again, it now DOES shows me the correct Downloads folder!

Go figure!

diegopradogesto, 2019-03-22:
Confirmed, it happens on macOS.

For internal reference CURA-6411. I already did a fix [here](https://github.com/Ultimaker/Uranium/tree/fix_save_path_macOS), let's see if that still works for the three platforms.

closed 2019-03-22 by diegopradogesto