Ultimaker/Cura #5469: Open project popup keeps appearing

regularlabs, 2019-03-21:
Using Cura 4 on OSX.

Every time I open a Cura project (.3mf), I get this popup:
[image]

As you see, the 'Remember my choice' option is checked. However, it seems to not do anything with that info. I keep getting that confirm popup.

Also in the settings, this option is saved correctly (but ignored):
[image]

And a connected other thing:
After clicking open, the 'Project Summery' is shown:
[image]

Can we disable that? As it just adds another irritating, unnecessary popup and click to the process.

Vandrasc, 2019-03-21:
Hi @regularlabs , first of all, thank you for your feedback!

Second, I am not able to reproduce the first issue, once checked the pop up doesn't appear for me anymore. Can you uncheck it and check it again? Maybe it is not updating correctly.

For the Open Project summary, I can bring it up with the team and maybe we can add an option in Preference where you can enable and disable it. We consciously decided to display it because it may override your current configuration.

regularlabs, 2019-03-21:
Ok, thanks.
Unchecking it in the dialog did nothing. It didn't even changelog the setting in the preferences.
I also rest that setting. And rechecked the option after restarting. But the dialog keeps showing everytime I open a project.

Please note: I open the project when Cura id not running. So double-clicking on the file, which starts up Cura.

Once Cura is running, the setting is working.

regularlabs, 2019-03-21:
Oh, scratch that last bit. Once Cura is running, the dialog appears too every time!

Ghostkeeper, 2019-03-22:
I'm also unable to reproduce this issue. It doesn't ask me again whether to open as project or import models after I checked the checkbox.

Do other preferences get saved properly after restarting Cura?

It does still ask whether to override the current profiles or create new profiles for them if I have any profiles in Cura that match the IDs of the profiles in the 3MF file, but have different settings. We discussed this, and we decided not to solve this. There are several problems with not showing that dialogue:
* The dialogue shows a choice for three profile layers. You could have a preference for each of them which is a bit esoteric, or you could have one preference that decides on all of them. But then that preference needs to be set to "ask first" before you get the choice for any of them. You can't "ask first but only if the material is different" then.
* If you click to override without asking, you'll get data loss without warnings, which is dangerous for professional users.
* If you click to always override without asking, you'll get hundreds of profiles at some point, which is not what anyone wants and slows Cura down.
* It's not worth the effort to find solutions for that.

regularlabs, 2019-03-22:
> Do other preferences get saved properly after restarting Cura?
The setting is saved. It just doesn't affect the behaviour.

> If you click to always override without asking, you'll get hundreds of profiles at some point, which is not what anyone wants and slows Cura down.
There are 2 options for the 3 settings: "Update" and "Create New". I guess you are referring to "Create New" when you say "override"?

Of cause, when hiding that summery, it would simply update the settings.
This doesn't save anything, it just changes the setting values, like what would happen if you would do this manually.

> If you click to override without asking, you'll get data loss without warnings, which is dangerous for professional users.
Then professional users shouldn't disable that dialogue!

Ghostkeeper, 2019-03-26:
> The setting is saved. It just doesn't affect the behaviour.

Sorry, let me rephrase: Do other settings get properly read after restarting Cura?

regularlabs, 2019-03-26:
> The first gets skipped when you set the preference.
That's the issue. It never gets skipped.

See below screencast:
[image]

Ghostkeeper, 2019-03-28:
I'm not able to reproduce that here. Here's what I'm seeing: http://dulek.net/work/saving_3mf_preference.ogv

Funny how the checkbox is enabled by default on your screen.

regularlabs, 2019-03-28:
Yeah, but you're not on Mac OSX.

Vandrasc, 2019-03-28:
I am using OS X and unfortunately, I still cannot reproduce. Did you have this issue with a previous version as well? 4.0 Beta or even 3.6? 

I am running Mojave 10.14 btw.

regularlabs, 2019-03-28:
Not sure if I had the issue on 3.6. Haven't used 4beta.
I'm on High Sierra 10.13.

closed 2020-06-20 by no-response[bot]