Ultimaker/Cura #5471: cx_freeze app data not found?

MightyMirko, 2019-03-21:
3.6/4.0
Win10 64bit
Ender3
Install Cura
Does not startup with and w/o administrative rights
Should start
Screenshot
https://imgur.com/a/QJ09UYA

Ghostkeeper, 2019-03-21:
Can you do this for me?

* Press Win+R.
* In the box, type in `cmd`. This should open a terminal.
* In the terminal type `echo %APPDATA%`.
* Respond in this ticket with what the terminal responds with then (perhaps censor your username again though).

MightyMirko, 2019-03-22:
Sure thing
C:\Users\[username]\AppData\Roaming>echo %APPDATA%
C:\Users\[username]\AppData\Roaming

MightyMirko, 2019-03-22:
C:\Users\[username]\AppData\Roaming>echo %PATH%
C:\opt\python371\Scripts\;C:\opt\python371\;C:\Program Files (x86)\SIEMENS\S7-PCT\bin;C:\Program Files (x86)\Common Files\Siemens\Sqlany;C:\Program Files (x86)\Siemens\Step7\S7bin;C:\Program Files\Common Files\Siemens\Automation\Simatic OAM\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Smart Projects\IsoBuster;C:\Program Files\PKWARE\pkzipc;C:\Program Files\PuTTY\;C:\Program Files\Git\cmd;C:\Program Files\Git\mingw64\bin;C:\Program Files\Git\usr\bin;C:\Users\[username]\AppData\Local\Microsoft\WindowsApps;%USERPROFILE%\AppData\Local\Microsoft\WindowsApps;

thopiekar, 2019-03-22:
Could you also show all your environment variables?
You can get the output via `set`.
I believe there is more ongoing on your computer 😉

Ghostkeeper, 2019-03-25:
It's apparently crashing on this line of code in cura_app.py:

https://github.com/Ultimaker/Cura/blob/fe66d15b9e1b615ac894daddf24e1a1aeecb2408/cura_app.py#L39

It should give a FileNotFoundError if it's unable to create some intermediary directory. In this case probably about permissions. In this case, it's expanding the user directory from the string `~/AppData/Roaming/cura`, which should expand to `C:\Users\[username]/AppData/Roaming/cura`. And Cura should have rights there. Maybe the folder is set to be accessible to administrators only?

MightyMirko, 2019-03-25:
```
Name                           Value
----                           -----
ADSK_CLM_WPAD_PROXY_CHECK      FALSE
ALLUSERSPROFILE                C:\ProgramData
APPDATA                        C:\Users\[user]\AppData\Roaming
AutInstLog                     C:\ProgramData\Siemens\Automation\Logfiles\Setup\
CommonProgramFiles             C:\Program Files\Common Files
CommonProgramFiles(x86)        C:\Program Files (x86)\Common Files
CommonProgramW6432             C:\Program Files\Common Files
COMPUTERNAME                   [hostname]
ComSpec                        C:\WINDOWS\system32\cmd.exe
DriverData                     C:\Windows\System32\Drivers\DriverData
HOME                           C:\Users\user]\1.gitconfig
HOMEDRIVE                      H:
HOMEPATH                       \
HOMESHARE                      \\[domain]\[user]
LOCALAPPDATA                   C:\Users\[user]\AppData\Local
LOGONSERVER                    \\[serv]
NUMBER_OF_PROCESSORS           4
OneDrive                       C:\Users\[user]\OneDrive
OneDriveConsumer               C:\Users\[user]\OneDrive
OS                             Windows_NT
Path                           C:\opt\python371\Scripts\;C:\opt\python371\;C:\Program Files (x86)\SIEMENS\S7-PCT\bin;C:\Program Files\Common Files\Siemens\Automation\Simatic OAM\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPo...
PATHEXT                        .COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC;.PY;.PYW;.CPL
PROCESSOR_ARCHITECTURE         AMD64
PROCESSOR_IDENTIFIER           Intel64 Family 6 Model 60 Stepping 3, GenuineIntel
PROCESSOR_LEVEL                6
PROCESSOR_REVISION             3c03
ProgramData                    C:\ProgramData
ProgramFiles                   C:\Program Files
ProgramFiles(x86)              C:\Program Files (x86)
ProgramW6432                   C:\Program Files
PSModulePath                   C:\Users\[user]\Documents\WindowsPowerShell\Modules;C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
PUBLIC                         C:\Users\Public
SESSIONNAME                    Console
Simatic_OAM                    C:\Program Files\Common Files\Siemens\Automation\Simatic OAM
Simatic_OAM_DATA               C:\ProgramData\Siemens\Automation\Simatic OAM
SystemDrive                    C:
SystemRoot                     C:\WINDOWS
TEMP                           C:\Users\[user]\AppData\Local\Temp
TMP                            C:\Users\[user]\AppData\Local\Temp
USERDNSDOMAIN                  []
USERDOMAIN                     []
USERDOMAIN_ROAMINGPROFILE      []
USERNAME                       [user]
USERPROFILE                    C:\Users\[user]
VBOX_MSI_INSTALL_PATH          C:\Program Files\Oracle\VirtualBox\
windir                         C:\WINDOWS
```

MightyMirko, 2019-03-25:
> 
> 
> It's apparently crashing on this line of code in cura_app.py:
> 
> [Cura/cura_app.py](https://github.com/Ultimaker/Cura/blob/fe66d15b9e1b615ac894daddf24e1a1aeecb2408/cura_app.py#L39)
> 
> Line 39 in [fe66d15](/Ultimaker/Cura/commit/fe66d15b9e1b615ac894daddf24e1a1aeecb2408)
>  os.makedirs(dirpath, exist_ok = True) 
> 
> It should give a FileNotFoundError if it's unable to create some intermediary directory. In this case probably about permissions. In this case, it's expanding the user directory from the string `~/AppData/Roaming/cura`, which should expand to `C:\Users\[username]/AppData/Roaming/cura`. And Cura should have rights there. Maybe the folder is set to be accessible to administrators only?
I tried both - installing with administrative rights first, then after reinstall without
And starting in compatibilty mode, with admin rights and as common user

fieldOfView, 2019-03-25:
I think https://github.com/Ultimaker/Cura/blob/fe66d15b9e1b615ac894daddf24e1a1aeecb2408/cura_app.py#L31 is not correctly expanding. If expanding ~ on Windows uses the combination of HOMEDRIVE and HOMEPATH, the folder would end up on @MightyMirko's H: drive.

cura_app.py should use os.getenv("APPDATA"), like resources.py does.

Ghostkeeper, 2019-03-26:
Let's give that a try then: https://github.com/Ultimaker/Cura/pull/5511

closed 2019-03-27 by LipuFei

Ghostkeeper, 2019-03-27:
We fixed something going by fieldOfView's insights. It still works fine for us here after the fix, but we couldn't reproduce the original issue in the end. Let's hope that it works better for Mirko as well.

Ellecross, 2019-04-03:
@MightyMirko I've a build for you that you can test. Can you try if the crash still happens for you with this? 
https://drive.google.com/open?id=1wi7QkAiBohea_syG4X2X9SuaWEg6rlEd

MightyMirko, 2019-04-04:
Hey @Ellecross i rqed your link

Vandrasc, 2019-04-04:
Hi @MightyMirko, you can use this build https://we.tl/t-TTqBMX2mpp.

Hopefully, this fixes your issue. 

Thanks.

MightyMirko, 2019-04-10:
@Vandrasc 
yeah this one worked. 
Thank you..

MightyMirko, 2019-04-10:
> 
> 
> Hey @Ellecross i rqed your link

Unfortunately, my gmail acc doesnt work anymore so i coudn't test your build

Ellecross, 2019-04-10:
> 
> 
> > Hey @Ellecross i rqed your link
> 
> Unfortunately, my gmail acc doesnt work anymore so i coudn't test your build

hey @MightyMirko no problem! It's the same build that @Vandrasc  sent you. :D 

Thank you for trying this out and good to know it works now for you!

PR #5511 (2019-03-27): Fix for weird home directories on Windows
The expanduser function will expand to a combination of HOMEDRIVE and HOMEPATH which might not be the same as APPDATA if your configuration is very weird (e.g. you've changed your APPDATA location).

Contributes to CURA-6418.
Hopefully fixes #5471.