Ultimaker/Cura #5472: UI slow to update

VjyedSC0UFMc8z, 2019-03-21:
**Application Version**
Cura 4.0.0

**Platform**
Windows 10 Education
Verison 1803
OS Build 17134.648
64bit

**Printer**
any

**Steps to Reproduce**
Edit cura settings by expanding print setting drop downs, or navigating list of settings under"Setting Visibility" in Preferences.

**Actual Results**
Menus take seconds or more to draw on screen when scrolling, expanding, hiding

**Expected results**
UI should be drawn as fast as I can click. 

Processor 
GPU is NVIDIA Quadro K600 with up to date drivers

Ghostkeeper, 2019-03-21:
We already did make the generating of the list of settings several times faster in 4.0. It's a continuous effort though.

There is a lot of things Cura needs to calculate for that list of settings. Each setting can have formulas to arrive at its value, such as the Outer Wall Speed being equal to `=speed_wall / 2`. To that end it tracks which settings are related to which. Also it has the same formula system to determine whether a setting needs to be made available, whether it is too high, too low, too high but just a warning, too low but just a warning and how to resolve global settings if multiple extruders have different opinions on what the setting value must be. It builds up a relation network to update these appropriately. Different levels of profiles can override setting values in different stages: Unsaved changes, saved profiles, quality levels, material profiles, nozzles, extruder definitions, global unsaved changes, global profiles, printer variants, machine changes and printer definitions. Of these, the material profiles, extruder definitions and printer definitions have their own inheritance structures.

All of that needs to be done for every of the 500 settings you have visible. In short, arriving at setting values and whether they should display errors or warnings or nothing is quite involved.

VjyedSC0UFMc8z, 2019-03-21:
Ok. Thank you for the update. I thought it was just hiding/showing already computed data.

closed 2019-03-21 by VjyedSC0UFMc8z