Ultimaker/Cura #5473: Supports to nowhere - new option minimum surface area to support?

slikk66, 2019-03-21:
CURA 4.0.0-BETA

https://imgur.com/a/jjpVciW

How to disable these tall thin supports that don't really seem to do anything?  Lots of times I get supports like these on tall objects that print all the way up (and usually fail) in tall thin towers, only to not really do anything.  Perhaps introduce a setting that needs to have a minimum surface area for a supported section?  Like unless the section is 2mm(square) don't bother with a support.

Thanks!

Vandrasc, 2019-03-22:
Hi @slikk66, in Support category you have a setting called [Use Towers], can you disable the setting and reslice? Are the support towers then gone? Thanks!

Ghostkeeper, 2019-03-25:
The minimum surface area to support is already implemented. It's new in Cura 4.0 in fact.

closed 2019-03-25 by Ghostkeeper