Ultimaker/Cura #5474: Cura 4.0 does not start - win 10

bebop-rocksteady, 2019-03-21:
**Application Version**
Cura 4.0 from https://ultimaker.com/en/products/ultimaker-cura-software - the recently released version. 

**Platform**
Windows 10 Pro

**Printer**
Cannot even add printer yet. Cura 4.0 is unable to start. This is the first time we install Cura on this PC. 

**Steps to Reproduce**
1. Start Cura

2. Cura does not launch

**Actual Results**
Cura is unable to start. I only see this message. 
[image]
The specifications of the PC is well above the minimum requirements as stated in your website. https://ultimaker.com/en/resources/20406-installation-cura-2-1
The video card drivers are updated to the latest as of today. 

**Expected results**
Expected Cura to start. 

**Additional Information**
Recently built a PC server in our office dedicated for slicing and printing. Me and some of us all get the same issue when we access this PC. But when we run the same version of Cura on our own laptops, Cura is able to start.

Ellecross, 2019-03-22:
Hi @bebop-rocksteady, nice name by the way. :D 

**TLDR**; It's not gonna work because the GPU in your remote PC does not allow remote rendering of Cura (uses openGL) via the RDP application you're using. 

I suspect that the PC server in your office is using an Nvidia Geforce GPU. 

I've seen this as well because we also have a test machine with Cura in it and when I'm accessing remotely, I get this same message. Thankfully, the test machine we have has both Intel integrated and dedicated GPU. When I need to access it remotely, I just disable the dedicated one and then proceed. 

**What you can do as workaround:** 
1. Try using a different RDP application like RealVNC or Teamviewer 
2. Use an integrated GPU instead on a remote 

I know it's frustrating, but it is a known limitation when trying to open an OpenGL application (like Cura) via RDP and you're using geforce GPU. 

I hope this helps you! 

Cowabunga!

bebop-rocksteady, 2019-03-23:
Thank you, this is good to know information. 

Yes we do have Nvidia Geforce GTX on our PC server. We do not have another option for now with our PC specification, unfortunately. 
I will report this back with my team and we will try with another way to remote into it.

Ghostkeeper, 2019-03-25:
Is your server actually able to render an interface on its own? If you plug in a monitor, does it show a desktop and such? We've seen that Cura is unable to run if the server has no desktop environment on it (though that was an Ubuntu server without X11, maybe it's similar for Windows servers).

bttguy, 2021-05-31:
So there is nothing to do if I have this error? A few release back Cura worked(until 4.7), but now(4.8, 4.9) I get this error? So if it was working and my laptop could run Cura, now why could it decide to stop running it? I used Cura for almost a year without issue.

Ghostkeeper, 2021-06-08:
You could start Cura locally, @bttguy, rather than via a remote desktop.

micnetix, 2022-05-03:
Having this issue on my computer with 3080ti Cura 5 just refuses to open because of opengl probe error all versions of Cura work before just 5.0 does not work anymore dont know if there is a way to bypass this probe error would be nice because my machine is only few months old and all the other Cura versions work fine

fieldOfView, 2022-05-03:
@micnetix are you trying to run Cura over an RDP connection? If so, you may want to read this:
https://www.khronos.org/news/permalink/nvidia-provides-opengl-accelerated-remote-desktop-for-geforce-5e88fc2035e342.98417181

If you are not running over an RDP connection, then - waves hand - this is not the issue you are looking for.