Ultimaker/Cura #5476: Bounding box of group is too large after translation

BagelOrb, 2019-03-22:
**Application Version**
Current master, likely any version up till now.

**Platform**
Linux, likely any

**Printer**
UM3, likely any

**Steps to Reproduce**
1- Load multiple models
2- merge them
3- Ctrl-Shift select each model individually
4- Move each model in the group in the same direction
(Alternatively, perform step 3+4 for each model separately)

**Actual Results**
bounding box of group is too large
instead of the bounding box being moved it is expanded
there is still some point included in the bounding box which is not from nay of the models

I couldn't slice my models until I realized my bounding box was wrong.
The workaround was to select each model and move them toward the unused space of the bounding box.

**Expected results**
Bounding box is never bigger than my models

**Additional Information**
[image]
[merge_bounding_box_problem.3mf.rename.zip](https://github.com/Ultimaker/Cura/files/2996366/merge_bounding_box_problem.3mf.rename.zip)

rburema, 2019-03-26:
I could reproduce it by playing around, but not relyably so. I wonder what the deciding factor is? It's a bug anyway.

Ghostkeeper, 2019-03-26:
I could reliably reproduce the issue. I put reproduce steps in CURA-6416 but they are more or less the same as here.

I fixed the bug: https://github.com/Ultimaker/Cura/commit/ff99c4e45f5a88ab303740a46b6fca0a5b36b2df

closed 2019-03-26 by Ghostkeeper