Ultimaker/Cura #5480: 4.0 - pressing the slice button should provide immediate feedback

connerk, 2019-03-22:
**Application Version**
4.0.0

**Platform**
WIndows 10

**Steps to Reproduce**
Press the `Slice` `Save to File` or `Preview`  button

**Actual Results**
App hangs momentarily (not the problem) to spin up the associated feature

**Expected results**
*When the slice button is pressed, it should notify the user that it has received the click.*
standard UX would have the button become a darker shade of the same color

rburema, 2019-03-26:
Thanks for reporting! We're aware of this issue and will get to it in time.
Devs: CURA-6034

Liger0, 2019-04-03:
Also, if you press again the Slice button, thinking it hasn't clicked correctly, it works as a "Cancel slicing" instead, even though the button isn't that.

JCDGitHub, 2019-05-18:
on windows 7 when you click again the slice button, the same process starts all over again, and after a while Cura tells 'Unable to slice', and finally the slicing process begins (The first one or the second I Don't know)

Ghostkeeper, 2019-05-20:
Yeah, it's the same problem. After a few seconds, we assume that the slicing engine crashed before it's been able to give a response. The real problem is that generating the slicing instruction towards CuraEngine can be very slow with large models and with complex setting structures (such as multi-extrusion printers).

JCDGitHub, 2019-05-20:
Ok GhostKeeper, 

Anyway, even if it's rather long for Cura to prepare the slicing, if the button is disabled  at once, there are 2 improvements : 
1 it give the user the information that the first click has been taken into account
2 it avoids the crash for the 'multi-click' case

Ghostkeeper, 2019-05-21:
Yeah I think we should turn the button into a "disabled" state after clicking it.

Ghostkeeper, 2019-06-03:
In the next version the button will be disabled and show "processing" after you've clicked it.

closed 2019-06-03 by Ghostkeeper