Ultimaker/Cura #5482: Not so much a bug, but a suggestion.

pacwind3, 2019-03-22:
It would be cool if the software showed you which orientation is the front and back of the plate. Right now, I'm using glass that doesn't fit my whole printer bed.  I started 3 prints last night, that weren't oriented correctly due to, not knowing where the front of the bed was going to be.

fieldOfView, 2019-03-23:
There’s an axis indicator on the front-left of the buildplate for this purpose.

bebop-rocksteady, 2019-03-23:
Agree with @fieldOfView. I did not notice this issue with UM3 because the printer model shows the orientation. BUt yes axis indicator is present and useful.

pacwind3, 2019-03-23:
awww, thank you so much for that clarification. I DID notice that indicator. That'll remove some frustration.

Ghostkeeper, 2019-03-26:
I think the axis indicator is sufficient, but it's ideal to have a build plate model for all printers. That's a continuous maintenance process though and if it's missing for yours, I hope that we can get it added some time. For now I'll close this issue because it's working as expected and its working is acceptable.

closed 2019-03-26 by Ghostkeeper