Ultimaker/Cura #5484: Prusa i3 Cura 4,0 When beginning print, extruder nozzle does not go to center of bed, instead will start printing at lower left corner

RecycledRoadkill, 2019-03-22:
Too many issues to check for all problems, I'm old and cranky. Simply tried everything I could thing of in cura 4.0 to start at center and it didn't decided to try 3.6 and printer works properly. BTW I'm printing (in this case) with an ANET A8.

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

smartavionics, 2019-03-22:
OK, old and cranky person, for there to be any chance of being able to diagnose your problem we need more information. Please save the cura project (File -> Save), zip the .3mf file and attach the zip to this issue. Without that it's not easy to help. Thanks.

smartavionics, 2019-03-22:
BTW, I'm old(ish) and cranky(ish) too!

RecycledRoadkill, 2019-03-24:
I just went back to version 3.6 which works great. Thanks any way.

rburema, 2019-03-26:
I'm sorry, but as @smartavionics said, you have given us very little information to work with. If you do decide to go back to 4,0; as a workaround, maybe you can go into the machine-settings and put 'Origin at center' to true.

rburema, 2019-03-26:
Perhaps related to #5502.

Ghostkeeper, 2019-03-26:
Maybe it's missing the home command in it start-up sequence? The Ultimaker 2 has that too if you print via USB.

fieldOfView, 2019-03-26:
> maybe you can go into the machine-settings and put 'Origin at center' to true.

Surely you meant him to check if "Origin at center" is false, which it should be for an Anet A8 (or indeed for almost every printer that is not a Delta printer).

rburema, 2019-03-26:
@fieldOfView Sorry, I think I misread what they wanted/needed. Also I don't have that much knowledge of non-UM printers (which I hope to rectify sometime). I guess I could/should have loaded the Anet A8 profile to check if it was one of the 'centered' machines.

fieldOfView, 2019-03-26:
As a rule, only delta printers should have the "Origin at center" checked. Perhaps the wording needs to change.

cryoaura, 2019-03-28:
> 
> 
> As a rule, only delta printers should have the "Origin at center" checked. Perhaps the wording needs to change.

Hey fieldOfView:

I have a slightly different issue than this one, but for mine and many others' knowledge, could you share why it is that only deltas should have an origin at center? I was under the impression this merely had to do with where the particular printer's firmware assumes it's zero value is at and not whether it's cartesian or delta. This is not a rhetorical question, truly am curious on why this is a rule and maybe it well help others troubleshoot. The BIBO printer specifies to pick origin at center, maybe this is wrong?

Ghostkeeper, 2019-03-29:
> I was under the impression this merely had to do with where the particular printer's firmware assumes it's zero value is at and not whether it's cartesian or delta.

Yeah, you're correct. But practically it's the convention to put the coordinate origin in the front left corner. Only Delta printers don't have a front left corner, so they would then put it in the centre. Maybe the BIBO printer is wrong, but that's why Cura has this setting, it can deal with that.

GregValiant, 2023-09-14:
Can this be closed?

closed 2023-09-29 by github-actions[bot]