Ultimaker/Cura #5485: Could not connect to the Cura Package database

kepbrown, 2019-03-23:
**Application Version**
4.0-PPA

**Platform**
Ubuntu 18.10

**Printer**

**Steps to Reproduce**
Click Marketplace Button

**Actual Results**
Window displays Could not connect to the Cura Package database. Please check you connection.

**Expected results**
A list of Plugins and Materials?

**Additional Information**
[cura.log](https://github.com/Ultimaker/Cura/files/2999011/cura.log)

bebop-rocksteady, 2019-03-23:
I too have this problem @kepbrown . I found that a restart of Cura 4.0, then it's fine.
But I do not see 'please check your connection' message. Only empty marketplace. 
My ticket https://github.com/Ultimaker/Cura/issues/5494 maybe we have same issue but different appearance? I don't know.

Ghostkeeper, 2019-03-26:
Looks like they are two sides of the same coin. @bebop-rocksteady's issue happens when the connection state is indeterminate. @kepbrown's issue happens when Cura got a confirmation that the marketplace is not reachable. I think it's safe to move this discussion into #5494.

closed 2019-03-26 by Ghostkeeper