Ultimaker/Cura #5486: Dual Extrusion Brim Issue

Kiendeleo, 2019-03-23:
**Application Version**
4.0.0
**Platform**
Windows 10 Pro

**Printer**
Custom Printer

**Steps to Reproduce**
1. Have Dual Extrusion
2. Select extruder 2 for the brim
3. Slice File
4. Slicer uses Extruder 2's temperature with extruder 1 to make brim

**Actual Results**
Printer attempts to print brim with extruder 1 at extruder 2's temperature. 

**Expected results**
Should print with the correct extruder and settings

**Additional Information**
https://imgur.com/jEkOjIo

Block137, 2019-03-23:
I think the root cause is SUPPOSED to be fixed sice beta here: https://github.com/Ultimaker/Cura/issues/5311
The slicer heat the extruder2 properly but does not insert extruder2 start code.

I'm also having this problem with installer downloaded 2 days ago so it's not fixed yet.

Ellecross, 2019-03-23:
Yeah, I recently tested the fix that @Ghostkeeper did for this. 

It's not in 4.0 yet, sorry! :( 

But it will be in the next one! Thanks for the report! 

... and good memory, @Block137 ! :) Thanks for linking the related issue.

Block137, 2019-03-24:
I saw it was fixed in beta version so I assumed it is already implemented in 4.0 release.
Can you give us some idea of fixed version release date?

Ellecross, 2019-03-25:
There is no fixed or final version release date. But I would say, lookout for it around June this year. :> Wheee

smartavionics, 2019-03-25:
Hello @Block137 , the latest release of my master branch builds (Linux and Windows only) contains that fix. If you wish to try it out, go to https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

You can install my builds alongside the Ultimaker releases.

Ghostkeeper, 2019-03-26:
Closing this as duplicate of #5311 then.

closed 2019-03-26 by Ghostkeeper