Ultimaker/Cura #5487: [4.0] UI issue in printer dropdown

muesli, 2019-03-23:
**Application Version**
4.0

**Platform**
Linux

In the printer dropdown, there are two links displayed on top of each other:

[image]

Ghostkeeper, 2019-03-26:
I can't reproduce this. Are you using the AppImage of Cura 4.0.0, or are you using a non-official release?

DoXer67, 2019-04-09:
I can confirm the issue. Running LINUX ARCH with KDE (uptodate)

nallath, 2019-04-11:
Are you using the appimage or using the build provided by the arch maintainer?

DoXer67, 2019-04-11:
I use it from AUR, so the build provided by the arch maintainer.
https://aur.archlinux.org/packages/cura-git/

muesli, 2019-04-25:
Same here actually. 43" UHD. Fairly regular DPI.

Ghostkeeper, 2019-04-25:
Does this also occur in the [AppImage](https://github.com/Ultimaker/Cura/releases/download/4.0.0/Cura-4.0.0.AppImage)? If not, we may have to take this up with the AUR package maintainer.

muesli, 2019-04-25:
@Ghostkeeper: yes, this only happens with the package from Arch, using Qt 5.12. The AppImage behaves fine in that regard.

DoXer67, 2019-04-26:
Same for me, AppImage is OK

Ghostkeeper, 2019-04-29:
I'm running from source as well using Qt 5.12 and it's not displaying here (Ubuntu 19.04)

[image]

arvind-d, 2019-07-22:
I have the exact same issues with the UI. I'm running Manjaro with KDE & QT 5.13 on a 27" 4K monitor, 1.5x scaling and 144DPI font settings.

Running the AUR package I get weird UI issues such as the first application run wizard buttons disappearing. The AppImage works as intended only when setting QT_SCREEN_SCALE_FACTORS=1. If I attempt to run it directly I don't have positioning issues but the texts are all fuzzy and gunky. The AppImage version also doesn't integrate well with the system theme like the AUR version does (in the configuration window mostly, getting dark gray texts on dark backgrounds).

[image]

Interestingly though, when running the AUR package via `sudo cura`, everything works as intended and displays perfectly. So, given that the app is running under the same QT version and displaying correctly with sudo, does this still have something to do with the QT version or maybe it could be some other conflicting settings?

[image]

nallath, 2019-07-23:
Since this doesn't occur in the appimage, this issue can be closed.

closed 2019-07-23 by nallath