Ultimaker/Cura #5489: [4.0] UI display issues

muesli, 2019-03-23:
**Application Version**
4.0

**Platform**
Linux

Layout issues in print settings dropdown:

[image]

nallath, 2019-03-26:
What version of linux are you using? Are you using the app image? What drivers are you using?

fieldOfView, 2019-03-26:
As commented [image] @muesli uses an unofficial distribution of Cura on ArchLinux with Qt 5.12.2

Ghostkeeper, 2019-03-26:
I'm also using Qt 5.12.2 when running from source, at the moment, but I'm not experiencing this problem. But if it doesn't occur when using the AppImage I think we can close this one. Let's hope that some updates from the package maintainers are able to figure out what the difference is between the versions of their libraries and the versions that the AppImage is using.

closed 2019-03-26 by Ghostkeeper