Ultimaker/Cura #5490: [4.0] Labels in config dialog are illegible

muesli, 2019-03-23:
**Application Version**
4.0

**Platform**
Linux

Group labels for Material are illegible:

[image]

fieldOfView, 2019-03-24:
Duplicate of https://github.com/Ultimaker/Cura/issues/4637

muesli, 2019-03-24:
@fieldOfView I've actually seen that ticket (and commented there as well), however it's marked as closed by the original reporter. I don't think it's really fixed though, he merely remarked a (wonky, if I may say so) work-around himself. Should probably re-open that one and close this one?

bliser3d, 2019-03-24:
Some shortcomings in the dark theme, I tried to fix it in #5499.

Ghostkeeper, 2019-03-26:
We've had the intent for a while to change the preferences screens to be Cura-themed rather than system-themed. That would resolve this issue as well.

esopalumpa, 2019-08-14:
Comparing the image from this bug and the images from #6195, the situation got worse.
[image]
[image]

Ghostkeeper, 2019-08-14:
I don't think it got worse since we didn't really change anything here in that regard. But it'll depend on your system colours (and which part of those system colours Qt can read).

esopalumpa, 2019-08-14:
Possible.
But this is the standard “Breeze dark” theme from KDE, so nothing fancy or makeshift.

esopalumpa, 2019-08-26:
It get’s a bit better if -style windows is given as option.

berberic2, 2019-10-13:
Problem still exists in 4.3

MaxVerevkin, 2020-03-26:
Problem still exists in 4.5 and -style option only partially solve it

NovaViper, 2020-04-13:
Will this issue ever get fixed?

Ghostkeeper, 2020-04-21:
It's still on our planning, since it's really quite difficult/messy code in the material manager. It's sometimes in the way of making improvements there, so we'd like to refactor that and at the same time give it Cura's theming rather than system theming which would fix this issue.

jaggzh, 2020-05-01:
> this seems to fix the issue (on kde plasma 5)
> env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP /bin/cura

This works for me. KDE Plasma also.

NovaViper, 2020-05-02:
> this seems to fix the issue (on kde plasma 5)
> env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP /bin/cura

This works! I have KDE Neon, I also used Cura's app image instead of the bin. However it makes the menu use the light theme instead of dark theme; but it's better than nothing

Schokobecher, 2020-09-18:
Sadly only a partial workaround for me. Start and End Gcode are not readable, but if copy/pasted into another window are correct. I assume some issues with the font used

[image]

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

LightTemplar, 2024-11-15:
@GregValiant 
I think so. Version 5.8.1, appimage, "Ultimaker Dark" theme:
[image]

GregValiant, 2024-11-15:
Thanks.  I go ahead and close this.

closed 2024-11-15 by GregValiant