Ultimaker/Cura #5491: CURA V4

Ex-Pilot, 2019-03-23:
I have just downloaded CURA V4 but it crashes whenever I try to do anything.

I will reinstall V3.6.

fieldOfView, 2019-03-23:
I have read your actionable report.

Thank you for your feedback.

bebop-rocksteady, 2019-03-23:
I get no crash when download Cura 4.0. :D 

(Not related to this ticket, but I saw your name @fieldOfView on the plugin Sidebar GUI. I downloaded and I am using it for Cura 4.0 Thank you for your work on the plugin! I find it really useful and easy to use. Others in my team in our office also like it. 

But I have trouble finding disable Extruder 2 in Cura 4.0 when I have plugin enabled. I am using Ultimaker 3 extended. In the original 4.0 UI, it is in the middle drop down. Do you know how I can disable extruder with the sidebar gui plygin? Thank you.)

fieldOfView, 2019-03-23:
The Sidebar plugin has its own issue queue here:
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues

In the sidebar, next to the extruder tabs, there's a dropdown/switch that lets you expand the configuration switch. That should also include the "Enabled" switch.

Ellecross, 2019-03-23:
@Ex-Pilot sorry that it crashes for you. 

Can you tell me more about the problem so that we can fix this and improve Cura? 

What error message do you see? Do you happen to have log files saved? 
or maybe you can send me your 3.5 configuration folder so that I can try to reproduce this issue. 

All these should be in %i\AppData\Roaming or if you go to Cura menu > HELP > Show Configuration Folder. 

Thanks!

closed 2020-06-20 by no-response[bot]