Ultimaker/Cura #5493: Lost focus

Ex-Pilot, 2019-03-23:
In V4 You have focused on data gathering instead of the actual purpose with this software. What a shame. 

V3.6 worked perfectly. My anti-malware warns me when opening V4. This did not happen in V3.6 or in any previous versions.

Please get back back to basics. I have used CURA for a long time and it has until now performed well. This version however is crap.

bebop-rocksteady, 2019-03-23:
You can add an exception for Cura 4.0 on your anti-malware application. It can happen sometimes unfortunately. And it does not mean that it's always true. It also happen to me when getting free CAD or CAM software online. But its fine, not bad. 

I think Cura 3.6 also collect data too, so not just 4.0. I remember seeing that message also in 3.6. GOod luck!

fieldOfView, 2019-03-23:
Also, you can turn off all data collection and update checking. See the privacy options on the General pane of the preferences.

ChrisTerBeke, 2019-03-23:
Cura has had the same data gathering policy since version 2.7. Sometimes after the launch of a new version the virus scanners haven’t been updated with the latest list of trusted software get, it’ll go away automatically.

closed 2019-03-23 by ChrisTerBeke

nallath, 2019-03-26:
Actually; The data gathering has been in Cura since legacy (I think something like 14.07?). The exact data that we gather has changed a bit over time of course, but it hasn't fundamentally changed.

Cura being marked as malware has something to do with the way we communicate with the backend (or that's what we think). The people who write anti-virus software don't want to tell us how or why we are marked and occasionally (i presume automatically) remove us from the white list (forcing us to re send the application to get it whitelisted again, but they also don't tell us that they did this)

Ghostkeeper, 2019-03-26:
> My anti-malware warns me when opening V4.

I think that's because the application is newer and not downloaded often yet. Nothing significant changed with when Cura makes a connection to any server between 3.6 and 4.0, except if you're connected to an Ultimaker 3 or Ultimaker S5 via the cloud because then it'll send the print to Google and from Google to your printer, rather than directly to the printer.

> Actually; The data gathering has been in Cura since legacy (I think something like 14.07?).

13.03 introduced it: https://github.com/daid/LegacyCura/commit/806d891dc85aad843bcb7e47dd28b0e6445b8153