Ultimaker/Cura #5494: Cura 4.0 - Empty Marketplace - no plugins

bebop-rocksteady, 2019-03-23:
**Application Version**
Cura 4.0 

**Platform**
Windows 10

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
1. Open marketplace
2. Everything is there 
3. Download sidebar plugin and then restart Cura
4. I start a print and send to my ultimaker extended via network 
5. I lock my PC and leave for about two hours 
6. I come back and I wanted to try downloading a new plugin 
7. I open marketplace and it is empty like the image. 
[image]
8. I wait, dunno how long, maybe 10 mins but still no plugins

**Actual Results**
Empty marketplace - just white. No plugins and no materials. 

**Expected results**
Marketplace to show plugins and materials that I can download. 

**Additional Information**
After I restart Cura and try marketplace again, it was working and showing plugins and materials again.

bebop-rocksteady, 2019-03-23:
Also

bebop-rocksteady, 2019-03-23:
I accidentally closed my own issue sorry

Ellecross, 2019-03-23:
> I accidentally closed my own issue sorry

😆 I thought it resolved itself already.

@bebop-rocksteady Thanks for reporting it. I see this sometimes too and it's been reported internally in Ultimaker as well but it's hard to reproduce for some of us. 

I'll bring this up with the dev team on Monday.

Ellecross, 2019-03-25:
Hey @bebop-rocksteady can you check if you also lose connection on your printer when this happens? 
We just want to isolate another issue that we suspect may be related. 

Thanks!

Dim3nsioneer, 2019-03-28:
Same happened here yesterday (Win10). It works normal again today. I do not have the sidebar plugin installed. My Cura also looses connection to the Ultimaker account (maybe related?), so I had to login again this morning.

Ghostkeeper, 2019-03-29:
> My Cura also looses connection to the Ultimaker account (maybe related?)

Could be. We've also discovered that issue on this side. Something to do with authentication tokens expiring and not being refreshed in time from Cura's end.

kepbrown, 2019-03-30:
Closing and restarting Cura does not allow me to connect to the market place. Sill receive the "Could not connect to the Cura Package database." message. No sidebar plugin installed on my Cura either.

BuildingOutLoud, 2019-04-01:
I am also having this issue. When I click Marketplace, it flashes a "Could not connect to the Cura Package database. Please check your connection." message for a very brief moment (I only know the message because I screenshotted it before it went away) and then goes to the empty marketplace screen. I am logged in to my Ultimaker account.
[image]

nallath, 2019-04-01:
Could you share your logs?

pdzamore, 2019-04-07:
I have the same problem. I do not have the sidebar plug-in installed. Sometimes Marketplace is populated, sometimes it is empty. Restarting Cura does not help.

OlivierAime, 2019-04-09:
Same problem here, Cura 4.0 on Ubuntu 18.10

2019-04-09 10:05:07,153 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [188]: Error check finished, result = False, time = 1.4s
2019-04-09 10:05:08,214 - INFO - [MainThread] Toolbox.src.Toolbox._makeRequestByType [553]: Requesting packages metadata from server.
2019-04-09 10:05:08,217 - INFO - [MainThread] Toolbox.src.Toolbox._makeRequestByType [553]: Requesting authors metadata from server.
2019-04-09 10:05:08,220 - DEBUG - [MainThread] Toolbox.src.Toolbox._createDialog [225]: Marketplace: Creating dialog [Toolbox.qml].
2019-04-09 10:05:08,688 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/ScrollView.qml:234:9: QML Binding: Binding loop detected for property "value"
2019-04-09 10:05:11,870 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
2019-04-09 10:05:11,871 - INFO - [MainThread] cura.CuraApplication.closeApplication [547]: Close application
2019-04-09 10:05:11,872 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [417]: Shutting down cura
2019-04-09 10:05:11,887 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [295]: Attempting to kill the engine process
2019-04-09 10:05:11,887 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [301]: Killing engine process
2019-04-09 10:05:11,888 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [304]: Engine process is killed. Received return code -15

forevertheuni, 2019-04-19:
Same problem here. Ubuntu using the PPA

nailgilaziev, 2019-04-19:
Same.  Any workarounds for a moment?

Ghostkeeper, 2019-04-23:
Workaround is to log out of your Ultimaker account (iirc).

OlivierAime, 2019-04-24:
I wasn't aware that you needed an account to access the marketplace. Anyway, I created an account, logged in, logged out. Same result : empty marketplace

Ghostkeeper, 2019-04-24:
You don't. It's just that creating an account allows you to rate plug-ins on it.

And it happens to make the requests to that server with log-in credentials if you're logged in, which causes this bug.

Ellecross, 2019-04-24:
Hello everybody,

If you are still encountering this empty marketplace in Cura 4.0, FYI - Ultimaker is aware of this issue and we are working on a way to fix this. 

It may need some time to properly test it after implementation so stay tuned. Sorry guys, that you're all seeing this. :/ 

As a workaround, you can try: 
1. If you are logged in, please try logging out and logging in again. 
2. If not logged in, please try restarting Cura,

LongLiveCHIEF, 2019-04-24:
same issue here on Ubuntu 19.04.

My current hunch is that everyone experiencing this problem has python3 as the default python interpreter on their OS.

fieldOfView, 2019-04-24:
> My current hunch is that everyone experiencing this problem has python3 as the default python interpreter on their OS.

Could you elaborate your hunch?
When you install Cura on Windows (like the OP), and when running the AppImage (which is the only supported way of running Cura on Linux), Cura runs independent on the system python interpreter. Also Cura will not run on anything but Python 3.

OlivierAime, 2019-04-24:
I just checked and it works using the Appimage. Not using the PPA...

Ellecross, 2019-04-25:
Ultimaker Cura devs, please see CURA-6487. :D

nallath, 2019-04-29:
This is fixed for the 4.1 release.

closed 2019-04-29 by nallath

michalss, 2019-05-06:
Same issue

sqsezs, 2019-12-25:
in version 4.4.1 the problem still exists

nallath, 2019-12-30:
Okay, but you do realize we will need a bit more info to fix this, right? So could you post your log files?
I can't reproduce this on my machine (the plugins & materials load just fine)

Ghostkeeper, 2019-12-30:
Takes a while sometimes though. Give it a few seconds.

Ghostkeeper, 2021-11-05:
It's a funny effect. The hundred-and-a-bit plug-ins and material brands on the Marketplace should at worst be similar to the settings list, but the settings list loads in like 2 seconds (asynchronously).

I've been thinking that it's probably either the secondary request being made for the authors, or to download all of the icons for each package/brand.

Ugly101, 2021-12-09:
Was this Marketplace issue ever resolved? I am having the same issue on my desk top but not on my lap top.

Ugly101, 2021-12-22:
Thanks Ghostkeeper. Just so happens that my computer had some other unrelated issues and I had to do a hard reboot. After the hard reboot Cura connected to the plugin data base.