Ultimaker/Cura #5495: [4.0.0] Layer preview not showing after slicing, adjusting the layer view slider does nothing

si-hb, 2019-03-23:
**Application Version**
4.0.0

**Platform**
Windows 10 on Microsoft Surface Book

**Printer**
Ultimaker UM3

**Steps to Reproduce**
See video: https://drive.google.com/file/d/1htkrfeVRI2csgFSwxnWhzVfmIBRJdFXv/view?usp=sharing

You can see that the replay tool will show the layer slicing correctly, but adjusting the layer slider does nothing. The model will not display the slice layers.

**Actual Results**
See video above

**Expected results**
Preview should behave as it does in 3.x versions

**Additional Information**
Appears to be related to the model, as other models I have loaded work fine. I've attached the project exhibiting the faulty behaviour.
[Bushing.zip](https://github.com/Ultimaker/Cura/files/2999632/Bushing.zip)

Ellecross, 2019-03-23:
Hi @si-hb , 

Can you go back to layer view > then open the Color Scheme - Material Color > then make sure to enable Extruder 1? Then let me know if this solves your issue. 

[image]

Thanks!

si-hb, 2019-03-23:
Hi, yes it did thank you.

I am aware of the setting, which I had enabled for both extruders on the previous version of Cura and hardly touched. I suppose when I upgraded to 4.0 I had expected it to behave exactly as 3.X did with my settings, so I did not think to check this at all, especially since the layer slicing would show correctly when using the animation tool.

rburema, 2019-03-26:
Good :-) Closing this as fixed then. I there's any issue related to this, please feel free to reopen.

closed 2019-03-26 by rburema

Ghostkeeper, 2019-03-26:
I wonder how the check mark got disabled then.

fieldOfView, 2019-03-26:
Probably in a previous version of Cura, which Cura 4 inherited its configuration from. In the old version the legend would always be visible, making this less of an issue.

PS: I could mention here that the Sidebar GUI always keeps the legend open, for this reason.

Black6spdZ, 2019-04-09:
having same problem as OP but my extruder 1 was already checked.. absolutely nothing shows on the layer view.. x-ray is fine

Stoccarda, 2019-04-16:
Hi,
Im using Cura 4.0.0 under Fedora 29.
I do no more see the sliders to scroll throgh the layer in the preview after slicing.
[image]

In a AppImage beta from January it was shown correctly.
Any setting necessary?

fieldOfView, 2019-04-17:
There is some weirdness going on with the Sidebar GUI plugin in the screenshot. Could you try disabling that plugin (Installed tab on the Marketplace, then restart Cura).

Are you using the AppImage?

Stoccarda, 2019-04-17:
Thanks for the hints, they are back again:
[image]
There is some overlap of the boxes and the print settings have to closed to see all other silders and boxes.
I am using the AppImage 4.0.0 form the download page.

Ghostkeeper, 2019-04-17:
Yeah you can resize the Drückeinstellungen by dragging the bottom of the panel up a bit, if you want to see behind it.

mrmax67, 2019-04-27:
[CCR10_M12 x 6mm nut v0.zip](https://github.com/Ultimaker/Cura/files/3124268/CCR10_M12.x.6mm.nut.v0.zip)

**Application Version**
4.0.0

**Platform**
Windows 7 64-bit on Sony VAIO VPCZ122GX laptop (Intel Core i5 2.4GHz 8GB RAM 465GB SSD)

**Printer**
Creality CR-10

**Steps to Reproduce**
Selecting Preview/Layer mode for some objects (usually smaller ones)

**Actual Results**
No layers available for viewing, only a grayed out outline of model. Layer scroll bar scrolls but has no effect on layer view.

**Expected results**
Each layer's tool path should display and the displayed layer should be selectable using the layer scroll bar.

**Additional Information**
I tried reslicing with no effect. I checked Layer>Color scheme>Extruder select checkbox (per another issue report I read) and I don't seem to have Extruder selections available in my dropdown (the attached png shows a screenshot.)
[image]
I've also attached a Cura project file with the filename extension changed to enable uploading.
[CCR10_M12 x 6mm nut v0.zip](https://github.com/Ultimaker/Cura/files/3124269/CCR10_M12.x.6mm.nut.v0.zip)

mrmax67, 2019-04-27:
Just discovered that if I change from Draft Settings (0.2mm layer height) to Normal (0.15mm) or Fine (0.10mm), layers are displayed. When I change back to Draft Settings, layers once again do not display.

Ghostkeeper, 2019-04-29:
Your profile works fine on this end:
[image]

Maybe this was something that was fixed in version 4.1?

revpar1980, 2021-01-12:
I have Cura 4.8 and it will not show layers.  I have tried suggestions above and they don't seem to apply.  Please help!
[image]

Ghostkeeper, 2021-01-25:
@revpar1980 You don't seem to have any 3D scene at all, so it's most likely a rendering issue. Try using compatibility mode for layer view, as found in the preferences:

[image]

revpar1980, 2021-01-25:
@Ghostkeeper you were correct!  Thanks so much for the easy fix!  I can see everything in the preview now.

Ghostkeeper, 2021-02-04:
That basically means that your graphics drivers don't implement OpenGL 4 well.

rudiniemeijer, 2021-03-23:
I stumbled on a workaround. I am using Cura 4.8.0 on an Intel MacBook Pro 16" with macOS Big Sur 11.2.3. I also had a blank display when selecting the Preview mode. My MacBook Pro, as many MacBooks do, has two video cards, an integrated Intel and a faster AMD Radeon Pro. I assume both have their own driver. Normally, an application gets the use of one of those cards, depending on the needs. The faster card uses more battery power. When I insist that always the AMD Radeon Pro is used, the Preview mode in Cura is present. When I let the MacBook decide, it selects the integrated Intel and the Preview mode in Cura only shows an empty screen, as reported in this thread. I have not exhaustingly tested use cases, so you may need to restart Cura to get the Preview working after you have selected the faster video card.

To select the faster card, choose Apple menu  > System Preferences, click Battery, click the Battery tab, then deselect the automatic graphics switching checkbox. In macOS Catalina 10.15 and earlier, this setting is in Energy Saver preferences.