Ultimaker/Cura #5496: [4.0] UI DPI unawareness

muesli, 2019-03-23:
**Application Version**
4.0

**Platform**
Linux

The UI seems generally pretty unaware of the underlying display's DPI. Mixed with what seems to be hard-coded positioning instead of using anchors and alignments, this leads to the UI breaking in many, many ways.

I've already filed separate tickets for the first few issues I encountered, but by now I'm aware that this is broken throughout the entire UI and it shows in so many places, that I'm not even sure where to begin and end with opening new tickets.

[image]

[image]

fieldOfView, 2019-03-24:
Perhaps ypu can share more about your environment and how you run Cura?

What Linux (version and distro), how did you install Cura (note that the only officially supported way is using the AppImage), and what version of Qt do you use?

fieldOfView, 2019-03-24:
Other users have reported layout issues with Qt 5.12. The official AppImage uses Qt 5.10.

muesli, 2019-03-24:
@fieldOfView This is running on ArchLinux with Qt 5.12.2. I'll try the official AppImage and report back.

muesli, 2019-03-24:
Confirmed, the AppImage looks a lot healthier. Thank you!

fieldOfView, 2019-03-24:
Could you check/update/close your other issues related to ui issues too?

closed 2019-03-26 by Ghostkeeper