Ultimaker/Cura #5498: Cura 4.0 and previous version. Too thin extrusion width

St007, 2019-03-23:
If model has elements width lets say 1mm and extrusion with set to 0.45 mm then Cura makes one additional line with extrusion width 0.1 to fit 1mm of model's width. This make an issue on large models because 0.1 width on nozzle 0.5 mm means that large amount of material will run out itself and on next run with normal width there will be large (2-5 mm) areas without extrusion at all, because of the empty nozzle.
[image]

Possible solution - use dynamic width, increase extrusion width for all lines if model can't be filled with whole number of lines with normal width

**Application Version**
4.0

**Platform**
Win10 x64

**Steps to Reproduce**
slice any models like pyramid or cone with 100% infill

**Actual Results**
issue on photo (attached)

**Expected results**
no issue

**Additional Information**
[image]
[image]

smartavionics, 2019-03-23:
Hello @St007 , please save the cura project (File -> Save) and zip the .3mf file and attach to this issue so I can take a look. Thanks.

St007, 2019-03-23:
Attached two projects.
Part1 - issue showed on render
top_fun_box -issue on result print
[Part1.zip](https://github.com/Ultimaker/Cura/files/2999814/Part1.zip)
[top_fun_box.zip](https://github.com/Ultimaker/Cura/files/2999815/top_fun_box.zip)

smartavionics, 2019-03-23:
Hello @St007 , thanks for the files. I can't slice Part1 using my Cura, it just hangs up using 100% CPU so I need to investigate what's happening there. I can slice top_fun_box and I think the problem is simply that it is doing un-retracted long travels so I recommend that you set the min comb distance without retract setting to a value of 10 or 20. I would also recommend setting the combing mode to no-skin.

I don't think 100% infill is worth the time and filament. Personally, I would just use gyroid infill and enable the connect infill lines setting. I would have thought that 40% gyroid infill would be stupidly strong, you could probably get away with 30% or even less.

I will get back to you re Part1.

St007, 2019-03-23:
Slicing Part1 takes 3-5 minutes on Intel Core-i5.
For top_fun_box problem also exists bacuse if detail shell width lower then wall thicknes but large then at least 2 extrsuions width, lets say 2.1 of extrsuion width, so infill rule is not applied. On my observation for top_fun_box there is not unretracted long travel, because it is actually extruding but 10 times less as per usual line, and that is problem.

smartavionics, 2019-03-23:
Just a thought, I didn't check and I'm away from my computer right now. Are you using relative extrusion? On some printers that doesn't work well at low extrusion rates. It's still worth avoiding any un-retracted travels longer than 10-20mm to avoid losing nozzle pressure.

St007, 2019-03-23:
I'm using custom made printer with latest Klipper firmware. So, I do not know if relative extrusion is issue. But I can see small enought lines on render (and sometimes they are way too small so not rendered at all), this may cause a problem not depending to printer model, because material can leak by itself if extrusion rate is too small.

smartavionics, 2019-03-24:
I sliced Part1 and, again, it's doing loads of un-retracted travels so I think you need to avoid doing that.

St007, 2019-03-24:
Is it unretracted travel ? https://user-images.githubusercontent.com/5739480/54871221-6dec4500-4db9-11e9-8dfb-aebb5e7cb350.png

smartavionics, 2019-03-24:
No, in that image you are pointing at wall gap fill (I think), the unretracted travel will be the dark blue lines, retracted travel are the light blue lines.

St007, 2019-03-24:
Good, so the problem is wall gap fill. It is filled with too little extrsion width as I pointed in first topic

smartavionics, 2019-03-24:
As you wish. But you've still got loads of unretracted travels. If you don't care about those, neither do I.

St007, 2019-03-24:
I got your suggestion and will do changes in settings. But it is not solving the main issue too little extrusion width.

Ghostkeeper, 2019-03-26:
You could set the Minimum Wall Flow higher, to like 50%.

Rotating the part sometimes helps to get the gap filling technique to work better for thin walls. We're working on a solution for this problem. That would be a duplicate of #5110 and a few other bug reports laying around here.

St007, 2019-03-26:
Yes, I tried Minimum Wall Flow, it has no effect on the issue in this report.

Ghostkeeper, 2019-03-28:
Yeah I'll close this as duplicate then. Please continue discussion in #5110 as far as your issues haven't been brought up there yet.

closed 2019-03-28 by Ghostkeeper