Ultimaker/Cura #5500: Support generated on both side of a wall, while only one is needed

Matioupi, 2019-03-24:
**Application Version**
4.0.0

**Platform**
Win10 x64

**Printer**
UM3

**Steps to Reproduce**
Try adding support for a print with small writing engraving (not going all the way through the wall) on the outer wall (e.g. on a vase). Support may not be actually needd in that case, but this will demonstrate Cura behaviour.

**Actual Results**

Cura generates support on the inner wall while it's absolutely not needed on that sdie as the engraving is not going though the wall.

[image]

[image]

**Expected results**
I would have expected support generated only on the outer wall

**Additional Information**
Behaviour does not depend on the fact that there is a hole at the bottom of the flower pot as shown on the screenshots

Ellecross, 2019-03-26:
Hey @Matioupi , can you try disabling the "Use Towers" setting please? let me know if this removes it.

Matioupi, 2019-03-26:
Hi, setting "Support / Use Tower" wasalready disabled. I just tried with / without and it gives the same inner useless support parts

rburema, 2019-03-26:
Hi @Matioupi I've tried to reproduce it quick-like but couldn't. Could you attach the model? (Use 'Save', not 'Export', and then use zip to prevent github from blocking the file.)

Matioupi, 2019-03-26:
I should have done this from the beginning :-)
here is the STL file : [test_cura_support_inner_walls.zip](https://github.com/Ultimaker/Cura/files/3007710/test_cura_support_inner_walls.zip)

rburema, 2019-03-26:
Thanks!

Actually, we'd prefer .3mf since that also saves the settings. For example: If I just slice the .stl with the default settings, I get way less support and also the bug doesn't show up. Could you at least tell us what the support-settings are you use?

cheers, Remco.

Matioupi, 2019-03-26:
Sorry... here is the 3mf file.
I believe I used default settings, only activating support:

[UM3_test_cura_support_inner_walls_3mf.zip](https://github.com/Ultimaker/Cura/files/3008017/UM3_test_cura_support_inner_walls_3mf.zip)

rburema, 2019-03-26:
Thank you.
I see you're printing with PVA (on the support extruder of course), which uses a different method of generating support. In this case, it'll do just a horizontal expansion around each overhang, so this is expected behaviour in that specific case (dissolvable support).

Ghostkeeper, 2019-03-26:
The problem here is known, but really hard to fix. The support areas are being offset with a certain distance (3mm for PVA by default). This offset can make it end up on the other side of your wall.

Matioupi, 2019-03-26:
Could it be possible to have 2 horizontal expansion values ?
I understand this would not solve all cases, but maybe in some cases would allow to se a value for the inner side that does not result in this behavious ?

Ghostkeeper, 2019-03-28:
Yes, we did think of this as well. A fix would be to offset in steps of one line width since that is the minimum width of something that could block the support, and in between each time subtract the solid shape from the support. You would see that if you have a hole in your wall it would expand only through that hole and not anywhere else. However that would kill performance since for your 3mm PVA support expansion that would add 8 extra offsets and 8 extra polygon diffs per support layer.

Alternatively you could revamp the whole algorithm so that you could include the Support Minimum X/Y distance in this step size, which would reduce the number of steps by about half with the default settings. But that would require doing the support X/Y/Z distance routines in a different order such that you do that during the Horizontal Expansion step. I don't know what the impact of that would be, since it's rather complex there.

A lot of unknowns.

Ghostkeeper, 2022-01-28:
This has been observed to be caused by Support Towers as well: https://www.reddit.com/r/Cura/comments/sepzwt/printing_wierd_supports/

GregValiant, 2026-09-07:
I'm cleaning up these old reports and requests.  I'll go ahead and close this.  If the problem still persists in 5.13.0 then please make out a new report so it goes to the top of the pile again.

closed 2026-09-07 by GregValiant