Ultimaker/Cura #5501: Any suggestions , Cura 4.0

lwbsbde, 2019-03-24:
Cura 4.0.0
My printer have a dual Extruder.
1. Can the settings of both Extruder be displayed side by side  (without switching)?

2. The objects can no longer be moved freely with the mouse. Only with the help of the three arrows.

rburema, 2019-03-26:
@lwbsbde 
1. No, sorry :-(
2. I find that it works if I first select (let go) then do the drag. If this isn't working for you, it's a bug.

rburema, 2019-03-26:
In the latter case, please provide wich platform you are on (win, mac, linux), and which graphics card you use.

Ghostkeeper, 2019-03-26:
Also take care that you don't start dragging too close to the origin point of the three arrows. It's ambiguous then which of the directions you want to drag, so we've prevented that. I think we can make some improvements in that regard.

lwbsbde, 2019-03-26:
Yes. It works the way you described it. For small objects, I have to enlarge the view to make it work.

lwbsbde, 2019-03-26:
a simultaneous view of the settings of 2 extruders would be great.

fieldOfView, 2019-03-26:
Please open a new issue for that. Add a sketch.

closed 2019-03-27 by nallath

Ghostkeeper, 2019-03-28:
Made a fix for that: https://github.com/Ultimaker/Uranium/commit/1aa4fe1ee5a27ce6789cebfa39a0489e1ac9ad0a