Ultimaker/Cura #5503: Cura 4.0 and brims

yet-another-average-joe, 2019-03-25:
**Application Version**

Cura 4.0

**Platform**

Windows 10

**Steps to Reproduce**

Slice with brim activated

**Actual Results**

Brims are printed from the object to the periphery. This gives a bad adhesion between the brim and the part, even with a priming line (I print mostly large PETG parts). This ruined the first print I attempted, and I had to downgrade to 3.6

**Expected results**

Same as previous versions : from "outside" to "inside", or an option to allow printing brims the previous versions way.

**Additional Information**

Is this a bug or a feature ? Am I the only one ? Did I lmiss something ?

I subscribed to the official forum in order to discuss this issue, but the post was not validated and was never published... (or I missed something !)

smartavionics, 2019-03-25:
Hello @yet-another-average-joe , it would be great to see the settings and model that you are using so please save the project file (File -> Save), zip it up and attach to this issue. Thanks.

yet-another-average-joe, 2019-03-25:
Hi,

as I wrote, I downgraded to 3.6. I could only upload a 3.6 .3mf file, I doubt it will help...

Something that could maybe (?) be interesting : before I install it and attempt a first print with 4.0, I had 4.0 beta and 3.6 installed for a couple of weeks. I was not familiar enough with the new interface, so I did'nt print anything with it). After I installed 4.0 (release version), an error appeared. A common error saying that everything had to be reinitialized (I already had this error a couple of times, no big deal). I was on a hurry (24+ hours print). I did'nt save the project and settings. I usually don't as I allways adjust parameters every time in order to verify everything, all parameters being displayed. The model itself slices and print fine with Cura 3.6, IRL and in layers view. It did'nt with 4.0.

I could try to reproduce the problem in a virtual machine. Cura 3.6 + Cura 4.0 beta + install 4.0 release, and see what happens.

But for now, all I can do is upload the STL... Parameters : 0.4 nozzle, 0.2mm layers, adaptative layers (for the bottom fillets...), 1.2 mm walls top and bottom, supports everywhere, brim, 15% infill gyroid, 40mm/s. Most parameters where default ones and calculated ones

[E3D Workbench Enclosure v1.1.zip](https://github.com/Ultimaker/Cura/files/3002995/E3D.Workbench.Enclosure.v1.1.zip)
. 

Do you confirm that Cura 4.0 should NOT normally print brims from inside to ouside (from the part to the periphery ?

smartavionics, 2019-03-25:
Thanks for the model file. Without the settings it's hard to say why got the result you did.

> Do you confirm that Cura 4.0 should NOT normally print brims from inside to ouside (from the part to the periphery ?

Yes, it should print from the outside, inwards (it does for me!)

yet-another-average-joe, 2019-03-25:
I will attempt to reproduce the issue, and will let you informed whatever are the experimentations results.

rburema, 2019-03-26:
@yet-another-average-joe We could _try_ the 3.6 file (3mf) in the 4.0 install, see what happens: If it reproduces, we're a step further, if not, we won't really lose anything.

yet-another-average-joe, 2019-03-26:
I generally don't save/use 3mf files. I don't have this one. I don't even have the gcode, as I resliced a couple of times before I uninstalled 4.0 and reinstalled 3.6.

I am aware that this will not be very helpfull ! I completely forgot to save files as I first thought it was a new feature. Maybe, for now, it could be better to close the issue.

I just reinstalled Cura 4.0, and now brims are printed the usual way. I played with parameters I usualy play with (supports, adhesion and adaptative layers), but I could'nt reproduce the problem. When I have some free time, I will reproduce the configuration : virtual machine + Win10 + Cura 3.6 + 4.0beta, 4.0 install, 3.6 uninstall. Remember : I got a corrupted configuration after I installed 4.0 over 4.0beta, with 3.6 and 4.0, and had to reset. After 3.6 uninstall : still corrupted. Then I reset (I did'nt uninstall the beta version before 4.0,  maybe the problem was there ?). This is weird.

rburema, 2019-03-26:
Thanks for your time anyway.
I'm/we're not going to close this _just yet_, since it isn't resolved and there might be other people with the same problem. (If it's really one of those _weird_ things where you can only get this if you've previously installed the Beta or/and are on a virtual machine, we're going to need that information, and not buried in the closed tickets.)

yet-another-average-joe, 2019-03-30:
I attempted to get Cura to work in a Win10 Virtual Box machine. Unfortunately, it needs OpenGL 2.0, and the virtual video adapter does not come with a GL 2.0 driver (GL 1.1 only :  https://forums.virtualbox.org/viewtopic.php?t=82614 ). I have no spare real Win10 machine I can experiment with, and therefore I can't go further.

Since then, Cura 4.0 has alway sliced normally.

Fips11, 2019-04-07:
Hello, I had the same problem. Cura changes the brim printing direction if you activate the option "Brim nur an Außenseite". If I deactivate this option the print order is again from outside to inside and there is no gap between object and brim.

yet-another-average-joe, 2019-04-07:
Save your 3mf file showing the problem !!!
I'm "happy" someone else has this problem, as I spent hours attempting to reproduce it, without success, and I was really thinking I had hallucinations !
It is not this option alone that causes the problem. There's some parameters combination.

smartavionics, 2019-04-07:
Hello @Fips11 , thanks for the report, I looked at the code and can see why that is happening.

Fips11, 2019-04-07:
Once you've found the bug, the 3mf file is no longer needed, is it?

smartavionics, 2019-04-07:
Sorry folks, I thought I had found the problem but no. So if someone can post a 3mf project file that exhibits this bug I will investigate further.

Fips11, 2019-04-07:
[CFFFP_ptfe catrriagev2.zip](https://github.com/Ultimaker/Cura/files/3051542/CFFFP_ptfe.catrriagev2.zip)

smartavionics, 2019-04-07:
OK, I think the bug is caused by having support brim enabled and there is some support brim inside the model (like in the holes in the example 3mf just posted). So the immediate workaround is not to have support brim enabled.

smartavionics, 2019-04-07:
I have a fix which I will be submitting as a PR later. I will also put that fix into my next Cura releases.

yet-another-average-joe, 2019-04-07:
Happy you got it !
It seems  this has been reported with another issue... #4829 
See the post by krz4tw (end of page)
Now I remeber I was playing with brims and supports when I had the problem.

landostolemycar, 2019-04-07:
I'm having the same issue with the support brim printing before the overall skirt. Disabling the support brim does get rid of the issue. Thanks for the help. Here's another 3mf with the same issue if it helps confirm your findings. 
[MSMV2E3D_MPSMV2_FilamentGuideToolHolderCombo - Copy.zip](https://github.com/Ultimaker/Cura/files/3051908/MSMV2E3D_MPSMV2_FilamentGuideToolHolderCombo.-.Copy.zip)

smartavionics, 2019-04-07:
Skirt gets printed before any brim..

[image]

closed 2019-04-10 by nallath

landostolemycar, 2019-04-10:
[image]
I must've uploaded the fixed one by mistake. Is enable support brim on?

rburema, 2019-04-10:
Hi @landostolemycar We've closed this issue because we've accepted @smartavionics fix, which will be in 4.1 (in absence of a 4.1 branch, merged to master, and pending QA review & perhaps systems testing, but I forsee no problems).