Ultimaker/Cura #5504: [4.0.0] Can't slice new object after loading gcode

dastumer, 2019-03-25:
**Application Version**

4.0.0
**Platform**

macOS 10.14.3
**Printer**

Creality Ender 3
**Steps to Reproduce**

Drag gcode file into Cura and wait for it to generate the preview. Then, drag an STL into Cura to create a new project.
**Actual Results**

The new object is visible in the prepare screen, but you are unable to slice it. The box where the option is is stuck saying 0 minutes, 0 material, Save to File.
**Expected results**

You should be able to slice the new object as normal.
**Additional Information**

[image]

rburema, 2019-03-26:
Hi @dastumer Does this happen every time for you? I've tried to reproduce this, but I could only get it to (not) 'work' once. (That doesn't mean it's not a bug, just that it'll be harder to fix...)

Ghostkeeper, 2019-03-26:
I could reproduce this also the first time only. It puts this error in the log:

```
2019-03-26 17:39:00,697 - ERROR - [Thread-1] UM.Logger.logException [84]: Exception: Job <UM.Mesh.ReadMeshJob.ReadMeshJob object at 0x7fe9acb624e0> caused an exception
2019-03-26 17:39:00,700 - ERROR - [Thread-1] UM.Logger.logException [88]: Traceback (most recent call last):
2019-03-26 17:39:00,701 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "/home/ruben/Projects/Uranium/UM/JobQueue.py", line 121, in run
2019-03-26 17:39:00,706 - ERROR - [Thread-1] UM.Logger.logException [88]:     job.run()
2019-03-26 17:39:00,733 - ERROR - [Thread-1] UM.Logger.logException [88]:   File "/home/ruben/Projects/Uranium/UM/Mesh/ReadMeshJob.py", line 40, in run
2019-03-26 17:39:00,750 - ERROR - [Thread-1] UM.Logger.logException [88]:     scale_factor_width = max_bounds.width / build_bounds.width
```

dastumer, 2019-03-27:
@rburema I'm able to reproduce it every time I quit Cura and reopen it.

rburema, 2019-03-27:
Sounds a bit like a value doesn't get initialized the proper way when this route is taken.

rburema, 2019-03-27:
Devs: CURA-6422

smartavionics, 2019-04-05:
Bug still in master branch today.

nallath, 2019-04-05:
Yup, that's because it hasn't been picked up yet. We've got about 90 open issues on our tracker, so it's taking a bit of time before we get to it.

smartavionics, 2019-04-05:
Sorry, I didn't word my comment well, all I was trying to say was that the bug exists in the master branch, I'm not complaining that it hasn't been fixed yet.

diegopradogesto, 2019-05-03:
I can't reproduce it today. Please let me know if some of you still can reproduce it with the master build, otherwise it was fixed somehow.

rburema, 2019-05-09:
@diegopradogesto No it still happens, I'm reviewing https://github.com/Ultimaker/Cura/commit/d5d37ccf0fdcc056f46611f6ed9ea719b66eceae now, which is supposed to fix it.

closed 2024-12-08 by github-actions[bot]