Ultimaker/Cura #5506: 4.0 - reloading model resized in external modeler does not always drop to build plate

shipbrook, 2019-03-25:
**Application Version**
4.0

**Platform**
Windows 10 Pro (Version 10.0.17134) 64-bit
Ubuntu 16.04.6 LTS 64-bit

**Printer**
Ultimaker 2+

**Steps to Reproduce**
1. Ensure that "Automatically drop models to the build plate" is enabled in the General Preferences dialog.
2. Create an STL which is too big to fit the build plate.
3. Load the STL into Cura.
4. Center the model on the build plate, if necessary, and verify that Cura renders it as unprintable.
5. Within the external modeling program, resize the STL so that it will fit on the build plate, and re-export.
6. Switch back to Cura and reload all models (either from the File menu or by the right-click context menu).

**Actual Results**
Cura renders the model as printable after the reload, but does not drop it down to the build plate. If the user doesn't rotate the view in order to notice that it hasn't dropped, this will cause the model to print on a significant platform of support (if support is enabled), or print as spaghetti due to starting unsupported in midair (if support is not enabled).

**Expected results**
Cura should drop the resized and reloaded model down to the build plate, if "Automatically drop models to the build plate" is enabled.

**Additional Information**
The issue also occurs if the model would have fit on the build plate, but was moved so that an edge was outside of the printable area, and was externally resized so that it shrinks enough to be within the printable area after the reload - it's rendered as printable after the reload, but not dropped down to the build plate.

If the model WAS rendered as printable before the external resize and reload, Cura will perform the "drop to build plate" as expected.

(Yes, I know people would normally just resize the model within Cura, and this issue would never be encountered. Ordinarily, I would have done that too, but the model I was printing had "sockets" for neodymium magnets, so I had to resize the model externally in order to preserve the diameter of the sockets.)

rburema, 2019-03-26:
Have reproduced this, can confirm it's a bug.

rburema, 2019-03-27:
Hi @shipbrook, Sadly, we've chosen to defer this for now. As you probably can imagine, there are many things that require our attention, especially after a release.

Vandrasc, 2019-08-12:
Fixed in https://github.com/Ultimaker/Cura/issues/6051, @shipbrook the fix will be present in the next release.

closed 2019-08-12 by Vandrasc