Ultimaker/Cura #5507: [4.0] GUI font size on Windows 10

uwezi, 2019-03-25:
**Application Version**
Cura 4.0.0

**Platform**
Windows 10 64bit

**Printer**
doesn't matter, but Wanhao i3

**Steps to Reproduce**
open the application

**Actual Results**
The font size in the settings - both with and without the GREAT sidebar plugin - is incredibly small. There is a lot of unnecessary white space around the text.

**Expected results**
Make the font size scaleable or at least use the Windows standard setting

**Additional Information**
see the screenshots from Cura 3.6 and 4.0 in comparison
[image]
[image]

prometx, 2019-10-23:
Seconded, the font size in this application is comically small. I have to lean into my screen to attempt to read. Can you guys cover my optometrist bill?

nallath, 2019-10-24:
Making the font size scalable (all the wile ensuring that the UI will still look good) is a whole lot harder than you'd imagine. 

That being said, if you're up for it, it is possible to change the theme files. If you have a look at the resources folder of Cura, you will find a "themes" folder. The "cura-light" is the default theme. In theme.json you will find a bunch of values regarding these sizes. Right at the top you will find a number of fonts, each with modifyiable sizes.

prometx, 2019-10-24:
Well, I have a *very* vivid imagination, and I'm just letting you know. I thank you though for your direction re: the .json file...

fieldOfView, 2019-10-24:
A better way than to edit the files in the resources folder is to make a subtheme. Go to Help -> Show configuration folder. The navigate to the "themes" folder there and make a subfolder named "larger". Download the following file and place it in that folder:
https://gist.github.com/fieldOfView/7de3ad07fa06afb821bc0c5fe0b668dc

Restart Cura, select the "Ultimaker Larger" in the Theme dropdown on the General preferences and restart Cura again.

Feel free to adjust the "size" values in the file to taste.

Ghostkeeper, 2019-10-25:
Our font size is dependent on the system's font scale. This can vary wildly depending on the OS settings (for accessibility etc) and screen DPI settings. And the rest of the interface should scale along for the most part.

That said, I believe our default font size is 95% of the system font size as to create slightly more contrast between important and less important information. Maybe we'd need to get rid of that.

fieldOfView, 2019-10-25:
> Our font size is dependent on the system's font scale.

Only partly so on Windows:
https://github.com/Ultimaker/Uranium/blob/master/UM/Qt/Bindings/Theme.py#L30

jonnyasmar, 2019-12-31:
> A better way than to edit the files in the resources folder is to make a subtheme. Go to Help -> Show configuration folder. The navigate to the "themes" folder there and make a subfolder named [...]

Went a tad bigger, switched to PT Sans for good measure, and based mine off cura-dark :) but I can actually read the GUI now. Thank you so much!!

okcoker, 2020-06-27:
Not sure if this is relevant to others but in my case, I noticed really small fonts for everything including the menu. Screenshot below:

[image]

I noticed this was because my main display is a 1440p monitor but I was opening Cura on my 2nd 4k monitor. When I make the 4k display my main monitor, the font size is decent as seen below:

[image]

Conversely, moving Cura back to my 1440p monitor makes everything look huge so it seems to me that part of the problem is that Cura doesn't recognize the monitor resolution I'm on.

Gamma36, 2024-06-22:
> Making the font size scalable (all the while ensuring that the UI will still look good) is a whole lot harder than you'd imagine.
> 
> That being said, if you're up for it, it is possible to change the theme files. If you have a look at the resources folder of Cura, you will find a "themes" folder. The "cura-light" is the default theme. In theme.json you will find a bunch of values regarding these sizes. Right at the top you will find a number of fonts, each with modifiable sizes.

I'm a retired developer. I suggest you consider SVG, which is one way for icons to scale from phones to TV sized monitors.

Also, beta testing is a good way to flush out accessibility issues. With 8K monitors in the pipeline you will eventually have to dynamically scale fonts like Chrome does. It can be done, just plan it in achievable steps.

closed 2026-09-07 by GregValiant