Ultimaker/Cura #5509: Staying logged in to Cura 4.0

DotNetRob, 2019-03-26:
**Application Version**
4.0.0

**Platform**
Windows 10 Pro 64 bit

**Printer**
Ender 3

**Steps to Reproduce**
Start Cura....

**Actual Results**
50/50 chance it will fail to login and tell me it can't connect to server.  As a results, can't access backups etc....   Have used cura 4.0 for a week and have had to login 3 times.  Cura 3.6 has never logged me out.

**Expected results**
Keep me logged in.....

rburema, 2019-03-26:
Hey @ianpaschal, this was the issue I asked Chris about in the app. I think he put in on your board? You might want to know the github issue so we/you/someone can close it when it's fixed (or you need more info from the user) :-)

DotNetRob, 2019-03-27:
[image]

opensourcefan, 2022-04-24:
4.13.1
Windows 10/64

Continues to sign me out, even if I don't close the app. Sometimes it signs me out so quickly, I'm still logged in on the webpage side.

nallath, 2022-04-25:
Please provide us with log files. There isn't a lot we can do to help you without it.

opensourcefan, 2022-04-29:
As an update I have found if I close Cura after using it, and then reopen it connects each time fine. 

If I leave the app open it will disconnect after a wake up from sleep or after a long period of time.

rburema, 2022-05-03:
@opensourcefan Extra information may be helpful, however, please just add the log files, they can be found from the help menu if you click 'show configuration folder' and then open `cura.log` in that folder.

closed 2022-05-23 by github-actions[bot]