Ultimaker/Cura #5514: Support macOS Fullscreen mode

Sanzenbacher, 2019-03-26:
**Application Version**
Cure 3.6.0

**Platform**
MacOS Mojave (10.14.2)

**Steps to Reproduce**
Press green button on top left of the window to go into fullscreen mode.

**Actual Results**
The App maximizes, like in windows.

**Expected results**
The App should go into the actual Fullscreen mode of macOS. It is much more convenient to work with apps in fullscreen on my macbook. Because this enables the three finger swipe gesture to switch fast from one app to another. I like Cura very much, this is the only complained I have. It would be great if it just worked as most of the other macOS app.

I understand that Cure is using QT, does QT have a problem with the fullscreen mode on Mac?

**Additional Information**
This is how it should work: http://osxdaily.com/2015/03/29/exit-full-screen-mode-mac-os-x/

vordenken, 2020-04-25:
Is there some kind of update to this?

Ghostkeeper, 2020-04-28:
Not really. It seems to have not been seriously considered, back in March 2019.

What happens when you click on View -> Toggle Full Screen?

vordenken, 2020-04-29:
View -> Toggle Full Screen actually works, it fullscreens the Cura application.

janikmdietz, 2020-05-01:
haha for the last four years or so every time I used Cura I wished for that feature. My life is worth living again I suppose :D

Ghostkeeper, 2020-05-07:
This seems to be something inherent in how the Qt framework creates windows. It can be customised, most likely, but I must say that I'm not entirely at home with how the maximise button is supposed to work on MacOS.