Ultimaker/Cura #5516: Multi-model settings for rapid setting testing

kfsone, 2019-03-27:
Use case: I'm planning to print 9 variations of a single small test model so I can see the impact of combinations of PLA and PVA settings (I'm getting a ton of stringing with my PVA), but THE most important tweaking parameters - temperature, cooling - aren't available cross-model.

None of the reasons for preventing them seem particularly technical.

Should it wait for the temperatures to be achieved? Yes. I *want* the model printed with those settings.

Where should it wait? If prime tower is enabled? There. If prime is enabled? There.

At worst case, it should wait just off of the last location it printed with a small z variance. If it drips onto your model, you shoulda enabled prime or blob.

There are all kinds of reasons that not having these is an issue: it's not guaranteed that the  printer/bed are leveled the same between prints - removing one print might offset that; it can significantly increase the time the whole process takes if you are not spending 24x7 sitting waiting for prints to complete.

You *can* use dual extrusion, but then you've added filament variance to the mix.

rburema, 2019-03-27:
@kfsone The technical reason for this is that some settings are set per extruder, and a lot of the engine-code relies on that. (For instance, most of the temperature settings (other than bed).) I know it still doesn't sound like it's a big problem from this description, but it would require a major refactor of the engine at least.

There _is_ a way to _sort-of_ do what you want, but unfortunately only for single extrusion, so testing PLA/PVA interfaces this way is not feasible: If the model is small enough (less than gantry height), you can enable one-at-a-time mode by changing the 'print sequence' setting that shows up when all but one of the extruders are disabled. Then a lot of settings can be tweaked per model that otherwise are inaccessible.

Ghostkeeper, 2019-03-28:
It's not just that this requires a major refactor. This was actually chosen explicitly for every of these settings.

If you want to print every object with a different temperature, what temperature should be used for objects that get merged with Union Overlapping Volumes? Or for support? The brim? Also, while waiting for your nozzle to heat up, you're going to be oozing and then you're going to need to prime again. So then you're going to need an additional prime tower for every object that you create. Where will that prime tower be placed? Can you be assured that this prime tower will not clip with any object and still fit onto the build plate?

If you still want it, you can use one-at-a-time printing in Cura 3.6 and adjust the temperatures per object. We removed it in Cura 4.0 though because it was buggy and never used, so you're on your own. Alternatively, you could use a custom FFF printer with 8 extruders and give each extruder their own temperature. Then add a Search-and-Replace post-processing script that changes all the T1 and T2 and T3's and so on to T0 and removes the pre-heating.

kfsone, 2019-03-29:
Thank you for the clarification, I appreciate your time detailing some of the parameters; and the 8 extruder idea is genius :) that works perfectly with the approach I have in mind since it explicitly involves N non-intersecting/touching copies of the same model. There are clearly going to be multiple constraints I haven't thought of, but does this feasibly sound like something that a plugin could do? (Arrange 4 copies of a model, change configs/temps between, ensure the model placements don't result in print head collisions during subsequent prints). Can the bed parameters be adjusted to prevent the head traveling anywhere in the half/quad of the other prints,  like a virtual bed?Sent from my Verizon, Samsung Galaxy smartphone

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Ultimaker/Cura/issues/5516#issuecomment-477645096",
"url": "https://github.com/Ultimaker/Cura/issues/5516#issuecomment-477645096",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Ghostkeeper, 2019-03-29:
> There are clearly going to be multiple constraints I haven't thought of, but does this feasibly sound like something that a plugin could do?

CuraEngine will not allow it. So the only way to do it would be to have your plug-in modify the output of CuraEngine after the fact. Since v3.6 (or was it 4.0?) CuraEngine will put markers in the g-code that indicate which object it's printing, so you could detect those in a plug-in or in a post-processing script, and put temperature commands before going to a certain object.

closed 2019-03-29 by Ghostkeeper