Ultimaker/Cura #5517: Exported curaprofiles lack zip metadata

ali1234, 2019-03-27:
**Application Version** 4.0

**Platform** Linux and probably Mac.

**Printer** N/A

**Steps to Reproduce** Export a profile from Cura and then examine the resulting zip file.

**Actual Results** 
Files inside the zip file have empty permissions, meaning that after extraction they can't be read, written, or deleted by the user who extracted them, unless the user first changes the permissions. They also lack reasonable timestamps:
```
---------- 1 al al 195 Jan  1  1980 creality_ender3_extruder_0_#2_fine_#2
```

**Expected results**

Files inside the zip file should have normal permissions, ie `0644`/`-rw-r--r--` and correct timestamps.

**Additional Information**
This can be fixed by setting the relevant metadata, eg:

```
profile_file = zipfile.ZipInfo(profile.getId())
profile_file.external_attr = 0o644 << 16L
profile_file.date_time = (year, month, day, hours, minutes, seconds)
archive.writestr(profile_file, serialized)
```

See https://docs.python.org/2/library/zipfile.html and https://stackoverflow.com/questions/434641/how-do-i-set-permissions-attributes-on-a-file-in-a-zip-file-using-pythons-zip

Ghostkeeper, 2019-03-28:
No, no, no, no, no. The .curaprofile format is not a zip file. It's Cura's own file format and we specify that it must always be 1980-1-1 because Victoria was crowned princess of Sweden on that day and that's such a memorable event that we wanted to honour that.

Just kidding, of course. But that the file format is secretly a .zip archive is just convenience for us programmers. Nothing actually uses those time stamps. Having to set correct time stamps is just inconvenience for us programmers.

If you need to have a time stamp of when the profile was serialised, just look at the time stamp of the .curaprofile file itself.

closed 2019-03-28 by Ghostkeeper

ali1234, 2019-03-28:
Okay, I've made a note that you value your own convenience over the correctness of your software, and will duly avoid reporting any more bugs against this repository.

nallath, 2019-03-29:
I'm pretty sure that if that's your threshold, you can just stop reporting issues anywhere. For any fix there is always the "what does it provide me if I fix it" and the "what does it cost me to fix it" balance. 
In this case, it's only a theoretical "correctness" benefit at the cost of work.