Ultimaker/Cura #5518: profile manager allows duplicate names and then gets confused about which profile is active

ali1234, 2019-03-27:
**Application Version** 4.0

**Platform** Linux

**Printer** N/A

**Steps to Reproduce**
1. Open profile manager and select any of the default profiles, eg "Fine".
2. Click on duplicate and enter the name "Fine".
3. Activate either the original or the duplicate, if it wasn't already.

**Actual Results**
Both profiles are marked as active in the profile manager, ie the activate and remove buttons are disabled for both profiles. In the side bar both profiles show a dot in the drop down. If the duplicate has custom settings they can be loaded by selecting it in the menu. If you play around with this for long enough cura crashes with a python exception.

**Expected results**
Either renaming a profile to a different name should not be allowed, or the profile manager should be able to better distinguish which one is actually active.

**Additional Information**

rburema, 2019-03-27:
@ali123 There's something going on  there: While I couln't replicate the scenario (but I'm on Win), it did crash the first time I tried to duplicate 'Fine' with the same name. Couldn't get it to reproduce afterwards. However, dissalowing duplicate profile names seems like a decent idea anyway.

ali1234, 2019-03-27:
The thing is that duplicate names are disallowed in some situations but not others. Here's a screencast demonstrating this and also #5519: https://youtu.be/rHb98UenZ04 (may have to wait a while before it is available in high resolution)

ali1234, 2019-03-27:
I've also noticed that Cura won't let me name a profile "High" even though no profile with that name exists. It is always renamed to "High #2", although I can create a profile with that name using duplicate.

LipuFei, 2019-03-27:
Made a fix for this: https://github.com/Ultimaker/Cura/commit/58507303fa6672ef71afb6f467a955b3baeba849

closed 2019-03-28 by Ghostkeeper