Ultimaker/Cura #5519: profile manager can be tricked into deleting the active profile

ali1234, 2019-03-27:
**Application Version** 4.0

**Platform** Linux

**Printer** N/A

**Steps to Reproduce**
1. Open Print settings side bar.
2. Open profile manager and create a custom profile. Doesn't matter how.
3. Activate the new custom profile - note that the Print settings side bar updates.
4. Rename the active profile. Notice it no longer shows in italics.
5. Click on a different profile in the profile manager but don't activate it.
6. Click back on the profile you just renamed.
7. Click "Remove".

**Actual Results**
The profile gets deleted but remains as the active profile in the print settings side bar. From here it is possible to crash cura if you mess around with settings with this invalid profile active (although I am not sure exactly how to reproduce that, I think it is something to do with either the update or create menu options.)

**Expected results**
Cura should never allow to delete the active profile, even if I rename it. Alternatively Cura should always allow it and then not crash.

**Additional Information**

rburema, 2019-03-27:
@ali1234 I couldn't reproduce this quickly, but see my comment on your last (probably related) issue, #5518. (It could also be that something got mixed up when you did the steps in your last report though.)

ali1234, 2019-03-27:
Yes, it seems that this only happens with profiles that contain a `#number`, whether it is put there automatically or you put it there manually. See screencast https://youtu.be/rHb98UenZ04

Ghostkeeper, 2019-03-28:
Fixed in Master.

closed 2019-03-28 by Ghostkeeper