Ultimaker/Cura #5522: [4.0.0] Layer thickness color view does not show different layer height infill etc

Block137, 2019-03-28:
**Application Version**
4.0.0
Also in 3.6 and possibly older versions

[image]
Layer height set to 0.06 mm and infill thickness set to 0.18 mm
You can see 6 layers of wall with only 2 layers of infill but they're all shown in blue (0.06 mm)

**Steps to Reproduce**
Use different layer height for walls/ infill/ support.
For example: 0.1 mm layer height and 0.2 mm infill
Use layer view with layer height visualizer option

**Actual Results**
Layer view show same color (same height) for all features

**Expected results**
Different line color for walls and infill when set to use different layer height

Ghostkeeper, 2019-03-29:
Yeah this is because the Infill Layer Thickness setting is actually not adjusting the thickness of the layer (not even partially). It just extrudes more material on one layer and then leaves out the infill in the layer(s) below. That's also why the lines appear much wider in layer view.

Fixing this is rather involved since we only store the layer thickness per layer, not per line. I'm going to defer this one since I don't think it's important enough to give this priority, though it is a valid bug.

nathanielstenzel, 2021-11-02:
I guess you still do not have this fixed in 4.11.0. For a little bit, I thought setting the infill height to double that of the layer height did not work because of this and then I decided to check the time and the single layer at a time preview and confirmed that the different layer heights do show but are not actually displayed correctly.

closed 2023-06-24 by github-actions[bot]