Ultimaker/Cura #5523: Please reintroduce Connect button on monitor page

Dim3nsioneer, 2019-03-28:
**Application Version**
Cura 4.0.0

**Platform**
Win10 Pro

**Printer**
UM3/S5

**Steps to Reproduce**
Go to monitor page and try to find the connect button for an unconnected UM printer

**Actual Results**
There is no connect button anymore

**Expected results**
For a not connected printer there should be a connect button again on the monitor page and not only a message to connect

**Additional Information**
please reintroduce the connect button as opening the manage printer dialog for simply re-connecting the printer after a new start of Cura 4.0 is not very elegant.

UltiArjan, 2019-04-06:
+1 ,  there's plenty of space for it :)

[image]

GregValiant, 2026-09-07:
I'm cleaning up these old reports.  I'll go ahead and close this.
I know this request has never been addressed.  There are just so many ways for USB printing to fail that it has basically been abandoned.  It is now disabled by default in Cura 5.13.0 and above.

closed 2026-09-07 by GregValiant