Ultimaker/Cura #5524: What I see is NOT what I get

Ex-Pilot, 2019-03-28:
I printed an STL file but what I got was not what I saw in Cura and in FreeCAD for that matter. Unfortunately I can't attach my STL file.

Without the possibility to attach the file it's difficult to explain my problem.

smartavionics, 2019-03-28:
OK, so what would be good is if you could save the project file (File -> Save), then zip up the .3mf file and attach the zip to this issue. Thanks.

Ex-Pilot, 2019-03-28:
[CCR10S_SS_TopDeck_V2.zip](https://github.com/Ultimaker/Cura/files/3017554/CCR10S_SS_TopDeck_V2.zip)

File attached.

In the final print I miss the two rings around the gun turrets. They should measure 1x1 mm. Instead I get a wall with a thickness of 2 mm.

smartavionics, 2019-03-28:
Thanks for the project file. Just sliced it and it looks like I would expect, a wall with a wider top and the required support. What's wrong with that?

[image]

[image]

Ex-Pilot, 2019-03-28:
I hate this. I didn't ask for support as it's not needed for this print but it must have been activated without my notice.

smartavionics, 2019-03-28:
Well, it's no good complaining to me, you need the support department (hahahahaha!)

Ex-Pilot, 2019-03-28:
Oh my God. I'm still laughing. Thank you for your help. The image you provided showed it all.

Support department!! Well, yes - of course.

Ghostkeeper, 2019-03-29:
As far as I can see from that 3MF file, support was activated manually by the user. It's not enabled from any of the profiles of that printer.

diegopradogesto, 2019-05-03:
Closing this issue then, since it's expected behaviour.

closed 2019-05-03 by diegopradogesto