Ultimaker/Cura #5525: supports density, Cura 4.0

jornada812, 2019-03-28:
strange behavior of the field density support. it is impossible to change the value if you do not change the distance between the support lines

Ghostkeeper, 2019-03-29:
Yeah that's because a higher support density can only be achieved by putting the support lines closer together. Your printer can't actually change the mass of the filament so it needs to put the filament closer together to change the density of the support infill.

We compute the support line distance from a combination of the line width and desired density using this formula: https://github.com/Ultimaker/Cura/blob/4f5aee629894e8d8f5dd2414affbb155d59a9f4b/resources/definitions/fdmprinter.def.json#L3886

closed 2019-03-29 by Ghostkeeper

jornada812, 2019-03-29:
It is beautiful, but intuitively incomprehensible. Disabled fields must be accompanied by tooltips or other instructions why the user can not change the value and how it can be changed. Otherwise, this leads to an opaque interface and lack of understanding in working with your program.

Отправлено из приложения "Почта" для Windows 10

От: Ghostkeeper
Отправлено: 29 марта 2019 г. в 12:55
Кому: Ultimaker/Cura
Копия: jornada812; Author
Тема: Re: [Ultimaker/Cura] supports density, Cura 4.0 (#5525)

Yeah that's because a higher support density can only be achieved by putting the support lines closer together. Your printer can't actually change the mass of the filament so it needs to put the filament closer together to change the density of the support infill.
We compute the support line distance from a combination of the line width and desired density using this formula: https://github.com/Ultimaker/Cura/blob/4f5aee629894e8d8f5dd2414affbb155d59a9f4b/resources/definitions/fdmprinter.def.json#L3886

nallath, 2019-04-01:
I've added an extra hint to the tooltip that should make this easier to understand. 
[image]

Liger0, 2019-04-03:
> 
> 
> I've added an extra hint to the tooltip that should make this easier to understand.
> [image]

I don't want to be a dumb, but "overriden" is missing a d.

Ghostkeeper, 2019-04-04:
That's not dumb. In fact, it seems some of us got this consistently wrong. Fixed here: https://github.com/Ultimaker/Cura/commit/bd1712cb10749914d79053426ff0397c1219b583

commit 64523c7b (2019-04-01): Add some extra info to the tooltip if a setting is not used due to the value of it's children

The behavior might make sense to us, but that's probably because we know what the behavior is. The
extra info in the tooltip should make it easier to understand

Fixes #5525