Ultimaker/Cura #5526: [Feature]Add object height in generated gcode

MB3Y, 2019-03-28:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Ghostkeeper, 2019-03-29:
This information is available in the Griffin header at the moment (if you were to slice for an Ultimaker 3 or S5). We could put it in the RepRap header as well, maybe.

closed 2019-04-01 by Ghostkeeper

Ghostkeeper, 2019-04-01:
It'll now be printed at the top of the file like this:

```
;FLAVOR:Marlin
;TIME:626
;Filament used: 0.111376m
;Layer height: 0.1
;MINX:37.2
;MINY:37.2
;MINZ:0.3
;MAXX:62.8
;MAXY:62.8
;MAXZ:10
;Generated with Cura_SteamEngine master
```

That is a 10mm cube. In the X and Y direction the cube is centred on the 100x100mm build plate, and there's a brim around it. In The Z direction you can see the height of the last layer and the height of the first layer.

MB3Y, 2019-04-04:
:thumbsup: :yum: