Ultimaker/Cura #5527: [Bug] Unable to extend menu settings area

Couby, 2019-03-28:
**Application Version**
Cura 4.0.0

**Platform**
Windows 10

**Printer**
Custom 3D printer - tried with multiple printers, no change.

**Steps to Reproduce**
Click on settings menu.
Try to extend the settings menu size, clicking the 3 dots maintaining the click and dragging.

**Actual Results**
The area of the menu is not extended. A slight move can be noticed but the size is freezed.

**Expected results**
The area of the settings menu should change accordingly the mouse drag.

**Additional Information**
A movie of the problem : https://youtu.be/BP6-LvL8wJA .
A possible factor ? the screen size resolution, set to 3840 x 2160 on this computer, haven't tried yet with another resolution more common.

rburema, 2019-03-29:
Strange. 1920x1200 is what I use on my main screens and it works fine (I'm also on Win10). Not sure what this could be. Just to be sure: I assume your drivers are up to date?

fieldOfView, 2019-03-29:
If anything, it is DPI related ("retina"), not actual screen resolution.

fieldOfView, 2019-03-29:
@couby, I noticed that the two resolutions you mention are not the same aspect ratio. Have you tried 1920x1080?

Couby, 2019-03-29:
@fieldOfView I don't have that resolution (1920x1080), but I tried other ones. That doesn't fix the problem.

As it works for others, sure it is not screen resolution itself but something related, maybe...

What do you mean by DPI ? What windows settings should I try to change to try identify the real problem ?

Couby, 2019-03-30:
Fixed.
Hard to clearly define where is the problem.
Yet I can describe what I've done.
I'm running Cura on my Laptop, and I use an additional screen when I'm at home.
Trying to move Cura on the second screen (1920 x 1080 resolution), all was too big so I closed Cura.
Running Cura again, it tried to launched on the second screen, but there I get Cura stucked unable to run. Shutting down second screen and running Cura again it started on first screen.
I played with the screen resolutions and succeeded running Cura on second screen. There the problem was fixed.
Moving Cura window on first screen, the problem was fixed too.
So I would say that seems related to using multi-monitor with different screen resolutions, but looks hard to reproduce.
I write this to help someone facing the same problem, in case of...

Ghostkeeper, 2019-04-01:
Maybe the window was too big to fit on the screen and that caused the window size to be some indeterminate faulty value? Since we use the window size to limit the menu settings area, if that's indeterminate the resizing might break.

I can't reproduce it that way though, but things might be different on Linux.

neophrema, 2021-04-18:
Hello there,
I seem to have the same issue, I cant resize the print dialog. The size is stuck at that mini-format: 

[image]

This is very cumbersome cause I need to scroll very slowly, have no overview.

Im on Linux using Sway as a window manager. It *might* be related but I actually dont think so. Im using the appimage of Cura.
Same behaviour on 4.7.1 and 4.9
Any idea?

fieldOfView, 2021-04-18:
You could try the SidebarGUI plugin

EverardoG, 2021-09-09:
I had the same problem that Couby described and fixed it the same way. It seems like this bug could be linked to using cura on both a primary laptop screen and an external monitor, unless neophrema wasn't also using an external monitor.

ejwahl2, 2021-12-19:
I have the same issue, but without the use of an external monitor. With the latest update (I was a few behind), I can no longer extend the settings window vertically beyond what is shown in the attached image. Previously, I could extend the settings window down to have much more visibility of settings options. I noticed slightly different amounts that I could extend when I changed aspect ratio and scaling options on my computer display options, but nothing that allows much further extension.

Let me know if any further information would be helpful!

[image]

tillig, 2022-03-07:
I also have this problem but the SidebarGUI plugin has worked around it for me. I agree with earlier comments that I think this has to do with DPI; I've seen similar challenges in other programs like Inkscape with higher DPI displays.

Ghostkeeper, 2022-03-20:
I have a high-DPI screen as well (a 4K beamer, configured to use 200% scale in Ubuntu) but I'm not encountering this issue on that screen either.

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

closed 2024-12-09 by github-actions[bot]