Ultimaker/Cura #5528: [4.0] Gantry height preset

Liger0, 2019-03-29:
**Application Version**
4.0

**Platform**
win 10 x64

**Printer**
Doesn't matter

**Steps to Reproduce**
Add a new printer

**Actual Results**
In the Gantry height field there are so many 9 that the mm unit measure is not even visible.

**Expected results**
The numbers and the mm should be kept separate

**Additional Information**
[image]

Ghostkeeper, 2019-04-01:
I've made this value default to the printer height. Makes sense anyway.

closed 2019-04-01 by Ghostkeeper

maukcc, 2019-05-10:
Unless specified in the definition file, the printer height is a good value.
At present it is printer height, regardless of what is in the definition file.

Ghostkeeper, 2019-05-13:
No, the printer definition can still overwrite it. Only the `default_value` gets hidden by the `value` now. Fixed that: https://github.com/Ultimaker/Cura/commit/e26e825d8318c70e4755a25f9b21b5f91293eca8

commit add6c0b4 (2019-04-01): Default the gantry height to be equal to the printer height

You can't fit any objects higher than that anyway.

Fixes #5528.