Ultimaker/Cura #5531: Question - How to clear out stale extruder files?

smartavionics, 2019-03-30:
**Application Version**

master (any)

**Platform**

Linux (any)

**Printer**

Not relevant

**Steps to Reproduce**

Load many different Cura projects that use different printer/extruder definitions. Once project is no longer used, delete printer.

**Actual Results**

Files related to the extruders defined in those project files remain.

**Expected results**

When a printer is deleted, the various extruder files that are associated with that printer should be removed.

**Additional Information**

The files shouldn't get left around when the printer is deleted. Alternatively some means of detecting and removing the stale extruder files should be provided. I have hundreds of files whose name starts with custom_extruder_.

smartavionics, 2019-03-30:
OK, so I removed all the files whose name starts with custom_extruder (having made sure all the printers had been removed first) and Cura appears happy.

Before I close this issue, does anyone have any comments re avoiding the build up of extruder files?

Ghostkeeper, 2019-04-01:
I can't reproduce it. Here's the two sets of reproduce steps that I tried:

1. Add Ultimaker 3 so that you have a back-up profile when deleting the other printer.
2. Add Custom FFF Printer.
3. Wait 10 seconds for the autosave. Verify that the files appeared in `~/.local/share/cura`.
4. Delete Custom FFF Printer.
5. Close Cura or wait another 10 seconds.

Now there are no files referencing Custom FFF Printer in either the `extruders` or the `definitions` folder. I also tried:

1. Add Custom FFF Printer.
2. Save a project file.
3. Close Cura and reset the configuration.
4. Start Cura and add an Ultimaker 3 to have a back-up printer after deleting the printer.
5. Load the project file.
6. Wait 10 seconds. Verify that the files appeared in `~/.local/share/cura`.
7. Delete Custom FFF Printer.
8. Close Cura or wait another 10 seconds.

Again, no trace of any Custom FFF printer any more.

smartavionics, 2019-04-01:
Thanks for putting time into that. It's not a big problem to delete the spurious files from time to time. I'll close this now.

closed 2019-04-01 by smartavionics