Ultimaker/Cura #5533: Spurious tool changes when using multiple extruders (4) and using support blockers

Schleichmichl, 2019-03-30:
**Application Version**
3.6 and 4.0

**Platform**
Windows 10

**Printer**
Custom Printer with 4 extruders

**Steps to Reproduce**
1) Create a custom printer with 4 extruders
2) disable the first 3 extruders
3) load a stl file that needs support
4) enable support
5) add a support blocker
6) generate gcode

**Actual Results**
;FLAVOR:Marlin
;TIME:5532
;Filament used: 0.00616596m, 0m, 0m, 1.18737m
;Layer height: 0.1
;Generated with Cura_SteamEngine 4.0.0
T3
M104 T0 S175 <------------------------------------
M104 S200
M105
M109 T0 S175
.
.
.
;Later on
T0 <----------------------------------------

**Expected results**
Only use and heat tool 3. 
There is no workaround possible (e.g. by commenting those lines during post processing) because by using tool offset in cura the coordinates are calculated using the wrong tool  !!!!!!!!!!

**Additional Information**
This also happens with custom support plugin. That being said, Cura can't be used to customize the support when having a printer with multiple extruders.
GCode flavor normally used is RepRap, but as i tried to reproduce it seems the bug is flavor independent.

Block137, 2019-03-30:
My fix:
1.Right-click the support blocker block(s)
2.Choose print with 4th extruder check box
3.Slice again

Worked on mine.

Schleichmichl, 2019-03-30:
Thx. It works as intended. Then there is a convinient workaround if you not it :-)

I would suggest that the default is adapted to the active support printing extruder.
That would completely solve this issue.

Ghostkeeper, 2019-04-01:
Yeah, seems like CuraEngine includes the support blockers in figuring out which extruders are used in the print.

LipuFei, 2019-04-02:
JIRA ticket: CURA-6440

Schleichmichl, 2019-04-04:
The workaround could be problematic when using multiple active extruders and  support blocker. 
Then, even selecting one extruder for the support blocker could be the wrong one.

Please add a testcase for such scenario :-)

Ghostkeeper, 2019-04-04:
Fixed [here](https://github.com/Ultimaker/CuraEngine/pull/1025) and [here](https://github.com/Ultimaker/Cura/pull/5556). A system test for this scenario was considered out of scope, but we're working on a structure to make testing that sort of end-to-end scenarios a bit easier.

closed 2019-04-04 by Ghostkeeper