Ultimaker/Cura #5535: [4.0] Double extrusion settings not hidden in single extrusion machines

Liger0, 2019-03-30:
**Application Version**
4.0

**Platform**
Windows 10 x64

**Printer**
Cr-10s

**Steps to Reproduce**
Use a preconfigured dual extrusion profile on a single extrusion machine

**Actual Results**
It crashes because it doesn't hide the dual extrusion options

**Expected results**
On cura 3.6 and older it reverted those options so you could still use the profile and it would auto-adjust it to work on the single extrusion machines.

**Additional Information**
[image]

Ghostkeeper, 2019-04-01:
Not able to reproduce this bug. These are my reproduce steps:

1. Add Custom FFF Printer.
2. Set the number of extruders to 2.
3. On the first extruder, change infill density to 21%.
4. On the second extruder, change infill density to 22%.
5. Save the current changes as a profile ("dual").
6. Add another Custom FFF Printer, but this time leave the number of extruders to 1.
7. Switch to the previously created profile.

Nothing crashes. The infill density shows 21% as per the first extruder.

Liger0, 2019-04-01:
Try using this profile (set for dual extrusion), and select as printer a cr-10s machine.
The dual extrusion settings will still be visible and Cura will crash when importing any stl.
[profile.zip](https://github.com/Ultimaker/Cura/files/3029722/profile.zip)

Ghostkeeper, 2019-04-01:
Ah, yeah it reproduces then. How did you create that profile?

Liger0, 2019-04-01:
> 
> 
> Ah, yeah it reproduces then. How did you create that profile?

Using a custom profile with 2 extruders.

Ghostkeeper, 2019-04-01:
That's what I did with the reproduce steps above, but that didn't reproduce the bug :disappointed:

Liger0, 2019-04-01:
I am able to reproduce it.
It even resetted automatically the machines and the materials.

1. Create a custom machine and set the number of extruders to 2
2. Assign different materials to each extruder
3. Set the "fine preset profile"
4. Set some thing like the "enable triggering tower", "enable support", "second extruder interface support"
5. Save the profile "Fine #2"
6. Load a cr-10s machine
7. Open an stl
8. Crash.

nallath, 2019-04-02:
The problem setting is the "support_interface_extruder_nr", which is set to the second extruder. Once you switch to the single extrusion machine, it can't find that extruder, because it doesn't exist.

Liger0, 2019-04-02:
> 
> 
> The problem setting is the "support_interface_extruder_nr", which is set to the second extruder. Once you switch to the single extrusion machine, it can't find that extruder, because it doesn't exist.

I still don't understand why it crashes and resets all the materials and machines. It should just give an error or deny the operation, not crash and reset everything.

nallath, 2019-04-02:
I think you didn't see that I closed it as part of the commit that fixed this ;) 

We totally agree that it shouldn't crash, which is why we fixed it right away.

Liger0, 2019-05-15:
I tried Cura 4.1 beta after a clean install.
It turns out this still hasn't been fixed.
[image]
Errors analisys:
`Traceback (most recent call last):
  File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 571, in _onStackChangeTimerFinished
  File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 684, in _updateDisallowedAreas
  File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\Settings\ExtruderManager.py", line 282, in getUsedExtruderStacks
KeyError: '1'
`

Register:
`Thread 0x00002674 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.1\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 84 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000033d8 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.1\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000031e4 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000016f0 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002ec4 (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000033f8 (most recent call first):
  File "X:\4.1\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 150 in _storeStderrToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002220 (most recent call first):
  File "X:\4.1\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 138 in _storeOutputToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000f04 (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.1\plugins\UM3NetworkPrinting\src\UM3OutputDevicePlugin.py", line 446 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000333c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.1\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.1\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap
[... 85 lines cut ...]
  File "X:\4.1\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x000021d8 (most recent call first):
  File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 306 in _logInfoWidget
  File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 156 in _createDialog
  File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 77 in __init__
  File "X:\4.1\build\inst\bin\cura_app.py", line 102 in exceptHook
  File "X:\4.1\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 341 in exec_
  File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 797 in run
  File "X:\4.1\build\inst\bin\cura_app.py", line 136 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
`
Note I have no such X drive.
I have C and D, and at this time Cura is on C, while in the past it was on D.

nallath, 2019-05-15:
What are the exact reproduction steps for you? You start a clean instance of Cura and then this immediately happens? Do you add a printer first? Do you open a project?

Liger0, 2019-05-15:
> 
> 
> What are the exact reproduction steps for you? You start a clean instance of Cura and then this immediately happens? Do you add a printer first? Do you open a project?

I load a cr10s machine and add a dual extruder custom definited machine.
I select the dual extrusion machine and create a profile with that. (see my old comment about it).
Then I try to select the cr10s machine and it instantly triggers the crash.

Liger0, 2019-05-15:
I also deleted the log to record the new one happening when I open cura and multiple crash report popup appear causing crash loop. Even restarting the PC doesn't solve it so I have to uninstall Cura.
This is the log opening cura just once:
`2019-05-15 17:57:00,764 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FileLogger
2019-05-15 17:57:00,783 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FirmwareUpdateChecker
2019-05-15 17:57:00,798 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FirmwareUpdater
2019-05-15 17:57:00,810 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeGzReader
2019-05-15 17:57:00,822 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeGzWriter
2019-05-15 17:57:00,834 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeProfileReader
2019-05-15 17:57:00,858 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeReader
2019-05-15 17:57:00,877 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeWriter
2019-05-15 17:57:00,892 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ImageReader
2019-05-15 17:57:00,905 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LegacyProfileReader
2019-05-15 17:57:00,919 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalContainerProvider
2019-05-15 17:57:00,932 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalFileOutputDevice
2019-05-15 17:57:00,944 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MachineSettingsAction
2019-05-15 17:57:00,957 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ModelChecker
2019-05-15 17:57:00,971 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MonitorStage
2019-05-15 17:57:00,985 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PerObjectSettingsTool
2019-05-15 17:57:01,000 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PostProcessingPlugin
2019-05-15 17:57:01,013 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PrepareStage
2019-05-15 17:57:01,025 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PreviewStage
2019-05-15 17:57:01,043 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RemovableDriveOutputDevice
2019-05-15 17:57:01,069 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimulationView
2019-05-15 17:57:01,090 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SliceInfoPlugin
2019-05-15 17:57:01,104 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SolidView
2019-05-15 17:57:01,118 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SupportEraser
2019-05-15 17:57:01,144 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [112]: No auth data to retrieve the access_token from
2019-05-15 17:57:01,148 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin Toolbox
2019-05-15 17:57:01,165 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CameraTool
2019-05-15 17:57:01,181 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MirrorTool
2019-05-15 17:57:01,199 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RotateTool
2019-05-15 17:57:01,217 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ScaleTool
2019-05-15 17:57:01,231 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SelectionTool
2019-05-15 17:57:01,250 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin TranslateTool
2019-05-15 17:57:01,273 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UFPReader
2019-05-15 17:57:01,291 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UFPWriter
2019-05-15 17:57:01,309 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UltimakerMachineActions
2019-05-15 17:57:01,416 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UM3NetworkPrinting
2019-05-15 17:57:01,440 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-05-15 17:57:01,444 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UpdateChecker
2019-05-15 17:57:01,480 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin USBPrinting
2019-05-15 17:57:01,504 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade21to22
2019-05-15 17:57:01,521 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade22to24
2019-05-15 17:57:01,538 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade25to26
2019-05-15 17:57:01,555 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade26to27
2019-05-15 17:57:01,572 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade27to30
2019-05-15 17:57:01,590 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade30to31
2019-05-15 17:57:01,607 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade32to33
[... cut ...]
2019-05-15 17:57:12,352 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
2019-05-15 17:57:13,768 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [188]: Error check finished, result = False, time = 2.9s
2019-05-15 17:57:14,543 - INFO - [MainThread] cura.CrashHandler._sendCrashReport [360]: Sending crash report info to [https://stats.ultimaker.com/api/cura]...
2019-05-15 17:57:15,953 - INFO - [MainThread] cura.CrashHandler._sendCrashReport [366]: Sent crash report info.
`

nallath, 2019-05-16:
I did the following:
- Delete all local files of Cura
- Add a CR10
- Add a custom fdm printer
- Set num extruders to two
- Change wall thickness of left extruder to 0.1 and the wall thickness to 0.2
- Create a profile with it
- Switch back to CR10 

I don't see a crash, however. What step are you doing differently?

Liger0, 2019-05-16:
I posted it above:

1.  Create a custom machine and set the number of extruders to 2 (use 1.75mm for nozzle).
2.  Assign different custom made materials to each extruder (1.75 materials).
3.  Set the "fine preset profile".
4.  Set something like the "enable triggering tower", "enable support", "second extruder interface support".
5.  Save the profile "Fine #2" and select it as active.
6.  Load a cr-10s machine.
7.  Open an stl.
8.  Crash.

nallath, 2019-05-16:
Even with your exact steps, I can't get it to reproduce.

diegopradogesto, 2019-05-16:
I think the only difference is the step 7. Does it happen when you load any STL or a specific one? In the last case, could you send the STL?

Liger0, 2019-05-16:
> 
> 
> I think the only difference is the step 7. Does it happen when you load any STL or a specific one? In the last case, could you send the STL?

I don't load any stl now actually.
In the past I had to do it, now it crashes when switching printer only.
Let's see if I can make you reproduce it..

1. Create a custom machine and set the number of extruders to 2 (use 1.75mm for nozzle).
2. Assign different custom made materials to each extruder (1.75 materials).
3. Create a Cr10S machine.
4. Be sure to select as active the dual extrusion machine.
5. Load my dual extrusion profile [PLA 0.4mm + PLA 0.5mm 0.16LH brass (Liger0).zip](https://github.com/Ultimaker/Cura/files/3187217/PLA.0.4mm.%2B.PLA.0.5mm.0.16LH.brass.Liger0.zip)
6. Switch now to the already configured Cr10S machine
7. Insta-crash.

nallath, 2019-05-16:
Right, I see now, It's doing more than your original reproduction steps. There is one setting in your profile that triggers the issue. 

I think i've got it fixed now (The crash reports also helped, so thanks for those :)) Linking the github issue in that field helped a lot.

Ghostkeeper, 2019-05-17:
For reference, I think this is the commit that Nallath is referring to: https://github.com/Ultimaker/Cura/commit/915e0e2047e466c17ae6754c9da1ba2a01166990

Ghostkeeper, 2019-05-17:
I'm unable to reproduce the crash with the dual extrusion profile. Seems that Nallath's commit fixed it then.

Liger0, 2019-05-29:
And.. it was *partially* fixed, but not totally. When adjusting the support settings and changing machines, the crash occured again. I sent a bug report via the dialog window with the link to github.

Ghostkeeper, 2019-05-31:
Found the crash you reported in our database:

Traceback (most recent call last): File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py",
 line 571, in _onStackChangeTimerFinished File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py",
 line 684, in _updateDisallowedAreas File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\Settings\ExtruderManager.py",
 line 271, in getUsedExtruderStacks KeyError: '1'

Ghostkeeper, 2019-05-31:
I've been clicking around, adjusting support settings and switching printers, but I can't get it to crash :(

commit efe8f191 (2019-04-02): Return the value for default extruder if the requested one could not be found

This fixes #5535
closed 2020-06-20 by no-response[bot]