Ultimaker/Cura #5536: [4.0] System tray icon Bug not fixed

Liger0, 2019-03-30:
**Application Version**
4.0

**Platform**
Windows 10 x64

**Printer**
It doesn't matter

**Steps to Reproduce**
Open and then close Cura

**Actual Results**
The icon in the system tray is still there until you move upon it.

**Expected results**
It should be fixed, like you report in the changelog as:
"Fixed an issue where system tray icons would remain when Ultimaker Cura is closed until mouse-over."
But it just isn't.

Ghostkeeper, 2019-04-01:
The PR that supposedly fixed this was merged into 4.0: https://github.com/Ultimaker/Uranium/pull/423

So maybe that didn't fix it completely. Can't test this on Linux though (OS prevents this anyway) so I'll ask one of our testers to try this tomorrow.

rburema, 2019-04-02:
Devs: New ticket made for this regression/issue is CURA-6439

Liger0, 2019-08-24:
Seems fixed for 4.2.1. Could you confirm that you've fixed it so it can be closed?

Ghostkeeper, 2019-08-26:
No, we haven't changed anything for this bug. Not on purpose, anyway. CURA-6439 is still open and I can't find any other tickets regarding to this apart from the original fix in November.

Liger0, 2019-08-26:
This is weird, because it doesn't happen with 4.2

rburema, 2019-08-28:
Could be fixed by a Windows update?

Liger0, 2019-08-28:
I think you're right. I retried on 4.0 and 4.1 and it isn't happening anymore. Weird. I did a clean install everytime and had this bug still on 4.0 and 4.1 at the time I posted this.

LipuFei, 2019-08-29:
I will close this issue then.

closed 2019-08-29 by LipuFei