Ultimaker/Cura #5537: [4.0] Multi-extruder improper temperature management

Block137, 2019-03-30:
**Application Version**
Both 3.6 and 4.0

**Printer**
Custom tool changer running Smoothieware

**Steps to Reproduce**
Slice a print that uses 2 or more extruders

**Actual Results**
Extruder2 activated then suddenly continue using Extruder1

Part of a result G-code
```
1. G1 F1200 E-11.5
2. M82 ;absolute extrusion mode
3. ;<T0END>
4. ;</T0END>
5. M83 ;relative extrusion mode
6. T1
7. M82 ;absolute extrusion mode
8. ;<T1START>
9. T1
10. G0 X433.0 Y369.8 F20000
11. M280 S5.4
12. G91
13. G1 Z3.5 F650
14. G0 X-40 F18000
15. G90
16. ;</T1START>
17. M83 ;relative extrusion mode
18. M105
19. M109 S170
20. M104 T0 S0 ;**This line**
21. M106 S102
22. G1 F1200 E5.5
23. ;MESH:40mmcube.stl(1)
24. G0 F13200 X0.00 Y10 Z5.36 ;**Also this**
25. G1 F300 Z5.36
26. G0 F13200 X176.131 Y212.369
27. M104 S180
28. ;TYPE:WALL-OUTER
29. G1 F300 Z5.16
30. G1 F1800 E4.5
31. G1 F900 X176.131 Y172.631 E0.51942
32. G1 X215.869 Y172.631 E0.51942
33. G1 X215.869 Y212.369 E0.51942
34. G1 X176.131 Y212.369 E0.51942
35. G0 F13200 X176.131 Y211.969
36. 
```

**Expected results**
Line 20 should be between line 4 and 5
Line 24 shouldn't exist (not related bug, more info at #5383 #2433 #980 )

**Additional Information**
In this example, T1 is active since line 6 (and continue till the end) T0 is no longer needed. But T0 cooldown line appears at line 20 (after T1 is activated), the controller read the T0 in that line and continue the print with T0 while actually holding T1.
_T0, T1 commands are "modal", when read by controller their effect will remain._

This is possibly firmware related.
But I couldn't find any documentation on how T command works.
[Smoothieware wiki](http://smoothieware.org/supported-g-codes) has the best definition I found.

> **Tool change**
> Tn will change the tool to n for all future commands, it may appear anywhere on the line or on a line by itself
> T1 M200 M200 T1
> 
> NOTE This is not compatible with the G-code spec, but unfortunately most slicers create incorrect Gcode for tool change.
> 
> Note in smoothie there must be a space in front of a line that is affected by a modal G code.
> Some G code processors like HeeksCNC don't prefix the space neither do they prefix each line with the modal G code.

Block137, 2019-03-30:
Found (ancient) duplicate: https://github.com/Ultimaker/CuraEngine/issues/466
Tldr, Marlin broke NIST G-code standard. Smoothieware adhere to the standard which nobody support. :(

Time for 3rd post processing script

Ghostkeeper, 2019-04-01:
It's not a standard if nobody adheres to it :wink:

I'm closing this as duplicate, but reopening that one. It somehow got closed but nothing shows who closed it or why.

closed 2019-04-01 by Ghostkeeper