Ultimaker/Cura #5538: Supports extending to bed, but un-needed

yet-another-average-joe, 2019-03-30:
Sorry, I had some problems while formatting this post, it is havoc !

**Application Version**

Cura 4.0, but the issue was there with 3.6 

steps to reproduce

- "walls" with "holes" (the technical terminology could be different, but I miss some vocabulary !
fillets around the bottom face
- supports everywhere

- Large holes not touching buildplate are sliced normally
- Small holes not touching buildplate have supports extending to buildplate

but the supports exist onlky between the part bottom and the plate, not in the part itself (hopefully !)

What should happen after the above steps have been followed

no supports between the part and the plate ; they are useless, and need some unwanted post processing

**Additional Information**

In this model, I trick Cura in order to have no infill in the sides but 100% infill (playing with nozzle diameter and part walls width), so the part walls are filled with straight lines (much more faster, and less stress for the printer). (later, I will have a feature suggestion about this)

It seems to be hole size dependant. I will model a part with different size holes and see what happens

[image]

[E3D Workbench Enclosure v1.2.3mf.zip](https://github.com/Ultimaker/Cura/files/3025813/E3D.Workbench.Enclosure.v1.2.3mf.zip)

yet-another-average-joe, 2019-03-30:
[update]

[supports01.zip](https://github.com/Ultimaker/Cura/files/3025990/supports01.zip)

[image]

[image]

[image]

And now, without fillet : 

[image]

PKav89, 2019-04-01:
It seems, like you have tree supports active with normal supports. Try disabling tree supports.

yet-another-average-joe, 2019-04-01:
Tree support is not activated (until now, I never used this function): 

[image]

Un-needed supports also appear when 'Touching buildplate" is selected :

[image]

Replacing filet with a 45° chamfer, therefore, it seems it is not related to the filet :

[image]

[image]

I am pretty sure taht this problem appeared with 3.6, and wasn't there before : I use filets and chamfers a lot, and did'nt see this before. Could some config file be corrupted ? Can you reproduce this on you setup ? (I uploaded a 3mf file in my previous post)

PKav89, 2019-04-01:
Sorry, my mistake. Try to disable "Use towers" and then, if it won't help, try to set Support horizontal expansion to 0.

yet-another-average-joe, 2019-04-01:
Bingo, it is towers related !

The glitches appear only when "Use Towers" is DISABLED and supports ENABLED (touching buildplate or not).

Ghostkeeper, 2019-04-01:
I consider this a bug. It's a duplicate of https://github.com/Ultimaker/Cura/issues/4219

The problem is that there is still support next to the model all the way down from the hole in your wall, due to the 0.2mm offset. This support is 0.2mm thick and thus too small to print. However down near the bottom of the model, the model gives way and then it is thick enough to print.

Closing this as a duplicate of #4219.

closed 2019-04-01 by Ghostkeeper